Jackysi/advancedtomato #154: Letsencrypt support for SSL certificates

WoLpH, 2016-04-13:
Especially when combined with the Nginx server this would be a very powerful feature, it would allow for anyone to use a proper SSL certificate without any warnings.

The big question is which client to use, the standard letsencrypt client is very bloated but simpler ones are available.

Letsencrypt: https://letsencrypt.org/

Jackysi, 2016-04-17:
Seriously what r u people expecting from a router :D. Please contact Shibby regarding this, because I only work on GUI and so I don't modify any system files. For this feature you would require firmware changes and so I can't help out.

closed 2016-04-17 by Jackysi

WoLpH, 2016-04-17:
Personally I think ssl is always a good idea. And in that case, having a
proper certificate is not a bad idea either :)