Jackysi/advancedtomato #155: Bandwidth Limiter for LAN1 (br1) guest Wi-Fi dont working on Cisco Linksys EA6700

SubZero77, 2016-04-14:
Hello!
I'v setup guest wi-fi network on Cisco Linksys EA6700 with 1.28.0000 -3.1-135 K26ARM USB VPN-64K and can't limit speed of guests wi-fi users (br1). They have full internet speed as normal users (br0)  Bandwidth Limiter for LAN1 (br1) don't working (( My settings screenshots:
http://i9.pixs.ru/storage/9/0/4/guestwifi1_9996191_21540904.jpg
http://i9.pixs.ru/storage/9/1/6/guestwifi2_6658751_21540916.jpg
http://i11.pixs.ru/storage/9/2/0/guestwifi3_2729056_21540920.jpg
http://i10.pixs.ru/storage/9/2/3/guestwifi4_3559753_21540923.jpg
Help me please!

sanbor, 2016-04-17:
I think I'm having the same issue with router Cisco E3200 with Tomato Firmware 1.28.0000 MIPSR2-3.1-135 K26 USB Mega-VPN

Jackysi, 2016-04-17:
Downgrade to Shibby's version 132 see this link on how to find it: https://advancedtomato.com/faq

closed 2016-04-17 by Jackysi

sanbor, 2016-04-17:
Sure! Thanks for your quick reply @Jackysi 👍

Jackysi, 2016-04-17:
You're welcome! :)

SubZero77, 2016-04-18:
Hello! After downgrade to Shibby's 1.28.0000 -132 K26ARM USB VPN-64K Bandwidth Limiter still does not limit the speed of guest users. Problem is not solved ((

sanbor, 2016-04-18:
Looking at your screenshoots you're not using bandwidth limiter properly. Here there is a guide. For my experience, there has to be a gap between each extreme. Download rate should be something like 200 and download ceil 2000. Here is a guide https://learntomato.com/setup-bandwidth-limiter-monitor-control-bandwidth/