Jackysi/advancedtomato #156: "nvram export" doesn't work in Tools -> System Commands

jluce50, 2016-04-16:
I'm not sure if this is an issue with the UI or if the command isn't supported. When I run "nvram export --set" to backup all my settings there is no output. Running something like "nvram get lan_ipaddr" does work however.

A related issue is that the output text is not easily selectable. It would be great if you could click on it and do Ctrl-A to select the output. As it is, Ctrl-A selects all the text on the page. Just a minor suggestion...

Jackysi, 2016-04-17:
I'm not sure but I doubt this is an issue. Feature wise I'm not sure that's really a useful addon due to many other stuff that needs to be taken care off first.

jluce50, 2016-08-14:
So this doesn't appear to be limited to System Commands. I'm getting ready to upgrade to a newer version with ad blocking and I have a number of settings I want to backup first. Since System Tools doesn't work, I tried executing "nvram export --set" via SSH and there is no output there either. I tried sending it to a file as well ("nvram export --set > settings.txt") and I get a 0 byte file. It looks like the command just doesn't work, period. Any idea why this might be?

jluce50, 2016-08-28:
I'm not sure if this is new or if I just noticed it. According to the tooltip on the System Commands page "nvram export --set" should work. As such, any chance this could be reclassified as a bug?

[image]

Jackysi, 2016-08-28:
This is still not a GUI issue. So this to report to Shibby and left out of AdvancedTomato issues tracker.

closed 2016-08-28 by Jackysi

jluce50, 2016-08-28:
Ah, okay. Apologies. Only reason I thought it might be a GUI issue is that it works on Shibby.