Jackysi/advancedtomato #160: R7000-AT-ARM-3.1-135-AIO-64K: Port FW, DMS and DDNS WAN IP address do not work

bhlevca, 2016-04-29:
I upgraded from v132 to tomato-R7000-AT-ARM-3.1-135-AIO-64K.trx.

As you mentioned this is an experimental build and indeed it is. I followed all the right install procedures but the firmware does not work in at least 2 aspecte that I found:

1) Port Forwarding does not work
2) DMZ does not work
3) in DDNS the Use IP WAN address always show s 0.0.0.0

I did the NVRAM erase. 

When I returned to 132  everything works again pointing to a 135 firmware issue.

Jackysi, 2016-04-29:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato and so its not actually a firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-04-29 by Jackysi

bhlevca, 2016-04-29:
What if the GUI is at fault because it does not pass the values properly? 
I suggest that you check that before dismissing the report.