Jackysi/advancedtomato #161: Tools are broken

bhlevca, 2016-04-29:
In 132 and 135 tested,  all the tools are not working.
I tested ping, traceroute and wireless survey.   The first 2 return :error no mater what you throw at them,

They all work in the command line, so the firmware from shibby is not at fault, it is a GUI issue.

System commands, I tried ls and ls -l and works, but this is pretty much what works. Commands that take arguments do not work. For example 'cd /tmp'

bhlevca, 2016-04-30:
One more thing, After running the wireless survey the WiFi is completely gone. Only a hard reboot fixes the issue. This is not happening on the stock tomato.

alindt, 2016-04-30:
I'm on 132 and everything works, no issues with WiFi after survey either.

[image]
[image]
[image]

bhlevca, 2016-04-30:
Alin, are you using R7000 build?

alindt, 2016-04-30:
no, I'm using R6250 and RT-N18U

bhlevca, 2016-04-30:
OK, different story then

alindt, 2016-04-30:
You should edit the issue title to reflect the specific model/build you're having problems with.

bhlevca, 2016-04-30:
You are right. I did that.  This was an addition to a bug I filed earlier

Jackysi, 2016-04-30:
I'm running R7000 v135 and every thing works just fine. Every thing under tools including trace, ping, system commands, WOL and Wireless Survey. I can't fix the issue if I can't reproduce it.

bhlevca, 2016-05-01:
I wonder what's the problem on my R7000 then? I am running v132 now because v135 is unusable because as stated in a different report Port  Forwarding DMZ and DDNS WAN IP don't work.

Therefore I can't test v135 at the moment. V132 looks OK for now as long as I don't touch the Tools.

The worst part for v132 is the WiFI Survey which breaks completely the Wifi and I can restore WiFi only by a hardware power cycle. Very weird.

Jackysi, 2016-05-01:
I have no idea, I'm not actually Tomato developer but more a web developer working on its GUI. My knowledge is limited to Graphical User Interface it self and not wifi, dmz, firewall and other problems you may experience. I have a lot of knowledge regarding Linux as OS and other software like NGINX etc, but they don't have much to do with Tomato. 

I can only ask you to try and clear NVRAM and if it still doesn't work contact Shibby regarding the specific issue. Sorry.

closed 2016-05-01 by Jackysi

bhlevca, 2016-05-01:
I already cleared the NVRAM twice. It did not help. I may try to install v136 tomato and see if I still experience the issues. 

Do you have  plans to upgrade to v136?