Jackysi/advancedtomato #162: [BETA] WAN IP won't renew on 134/135 releases

Alcolawl, 2016-05-03:
Router: ASUS RT-AC68P
Firmware: v3.1-134/135
Notes: I understand this is a beta/experimental release. I'm not begging for a fix, I'm simply filing this so the dev is aware of an issue.

Issue: WAN IP does not renew after upgrading to either available experimental build. This happens regardless of whether the nvRAM is wiped after the upgrade or not.
[image]

ruslan61rus, 2016-05-03:
You can just print in command line:

> nvram set wan_pptp_dhcp=0
nvram commit

> 
and issue will be fixed

Alcolawl, 2016-05-03:
Executed the commands and rebooted the router. Did not fix the issue.
[image]

Alcolawl, 2016-05-10:
Issue has been fixed. After comparing several screenshots and information I noticed that the MAC address being assigned to the WAN port was incorrect after flashing build 135 and clearing the nvram. I went to "Advanced Settings" > "MAC Address", changed the WAN port MAC address to the correct address and the problem was resolved.

closed 2016-05-10 by Alcolawl