Jackysi/advancedtomato #164: Wireless 2.4GHz/eth0 Settings Missing

dmushail, 2016-05-13:
Hi, 

Linksys EA6500v2
v3.1-136

After upgrading to v3.1-136 from v3.1-135, my 5GHz SSID wasn't showing up. After going to the "Status Overview" page, I saw that my 5GHz adapters SSID was renamed to the 2.4GHz's SSID and all settings for eth0(2.4GHz) have disappeared. There is no sign of it in Status Overview, Networking, or Advanced->Wireless. Interestingly, after renaming my 5GHz SSID back to the way it was, the 2.4GHz SSID didn't stop broadcasting and continues to work on my 2.4GHz only clients.

closed 2016-05-14 by dmushail

underley, 2016-09-10:
Almost the same issue with 1.28.0000 -3.3-138 K26ARM USB AIO-64K on Linksys EA6700
Via ssh I can see:
 LAN : 10.***.***.1/24 @ DHCP: 10.***.***.193 - 10.***.***.199
 WL0 : 2,4GHz @ ****** @ channel: PL6 @ 48:F8:B3:7C:**:**
 WL1 : 5GHz @ ****** @ channel: PLauto @ 48:F8:B3:7C:**:**
but ifconfig shows only eth0 (lan) and eth1 (WL1, 5GHz)

phcorrigan, 2016-09-12:
I'm having a similar issue. I just installed Advanced Tomato on a Linksys EA6500V2. If i change the DHCP address range, change the Country setting, or change the SSID for the 5 GHz band I lose the 2.4 GHz band when I reboot. I suspect other changes will cause this as well. I even tried only enabling 2.4 GHz, but when the router is rebooted it comes back as 5 GHz, only but adopts the SSID that I assigned to the 2.4 GHz band. I am using Advanced Tomato v3.3-138 AIO-64K. Has anyone seen this issue and, if so, found a solution? Thanks.

dmushail, 2016-09-12:
I don't know if this will help you, but what I did to solve the issue at the time was re-flash the router with the "Restore Defaults" option enabled. It was inconvenient to have to reconfigure everything, but I haven't had the problem since.

phcorrigan, 2016-09-12:
I'll give it a try. Thanks.

phcorrigan, 2016-09-13:
Reflash with Restore Defaults" didn't help. Thanks.