Jackysi/advancedtomato #165: Mobile experience

nemchik, 2016-05-15:
I read in the change log think that a responsive layout and a mobile friendly interface is on the books, but I didn't see an issue here to track discussion about the topic, so I'm opening this to hopefully see where it goes.

Currently as of 3.1-136 the interface on mobile doesn't let me access the sub menu to get to all the options under administration (I can only access the top few items in the sub menu). The html page height seems to be set really weird on mobile as well, leaving a lot of white space before the actual page itself. The pages are all zoomed in by default too.

Anyway I've just recently come back to this firmware for my main router and so far I'm happy with switching back, but I definitely hope to see improvements with the mobile interface.

Jackysi, 2016-05-16:
I agree, this will be updated in future its just I don't have time at the moment :(

Jackysi, 2016-05-30:
The responsive design has been severely updated (fixed). The "automatic" zoom is what "responsive" means. It means it adapts to device screen size and PPI. Since Tomato has some large inputs and boxes I decided to make different approach to mobile view. Basically the most changes are within navigation menu and specified content width to minimum 720px.

Try it out and let me know how u feel it (version 3.2-136 which isn't out yet)

nemchik, 2016-07-03:
3.2-136 navigation has been great! I've played with it a ton and I can access everything! Functionally speaking everything is usable, and it all looks better than Sonny's base.

The page content doesn't feel fully responsive though. I notice places that are two columns on the desktop are nicely compacted into one on mobile, but there is still a lot of sideways scrolling which seems to be the result of padding, margins, or text size in some cases.

I might be getting too nitpick-y (maybe the only person who routinely accesses my router from my phone). Anyway thanks for the overhaul on the menu.

Jackysi, 2016-07-03:
Its very hard to fit Tomato design into mobile screen and doing it perfectly takes a loot of time. But slowly things progress. I'm not sure what u mean about the "Sonny's base"? Is that some sort of modification or something? :) 

I'm happy you like the change and that you find it useful =)

nemchik, 2016-07-03:
Sorry I meant shibby! Stupid auto correct on my phone.

Like I said, I'm probably one of the few (only) people to frequently access my router from my phone on a frequent basis. I don't expect rainbows or fireworks, and I think you've done a great job with the interface, and the navigation fix might be the end of this discussion. If so feel free to close this ticket. If you intend to make further changes that you already have in mind then I'd love to discuss them or provide feedback here or in another ticket.

Jackysi, 2016-07-03:
Great! Thanks for report and yeah this will be worked on in future however slowly =)

closed 2016-07-03 by Jackysi