Jackysi/advancedtomato #168: Empty QoS Bandwidth Graphs

outlyer-net, 2016-05-21:
The QoS graphs for bandwidth are blank on my router, plus all bandwidth measurements below them stay at zero. The connections distribution graph is displayed correctly.

Router: Asus RT-N16
Interface version: 3.1 (-135 AIO)
Screenshot:
[image]

Jackysi, 2016-05-24:
Did you care to enable QOS?

outlyer-net, 2016-05-25:
QoS is enabled in _Basic settings_. However, there might be something else going in since the _B/W limiter_ page displays the warning that QoS is disabled.

Upon further tests, I've identified what makes the graphs disappear, at least for me:
Having a _DSL Overhead Value - ATM Encapsulation Type_ other than None seems to be the culprit.
Setting it back to None brings back the graphs.
The _B/W limiter_ keeps telling me QoS is disabled nonetheless.

Also, my router lost its settings when troubleshooting this, and I used that chance to update to 3.1-136, so I can't reproduce the exact same setup it had before, but the behaviour seems to remain the same.

Jackysi, 2016-05-29:
That's strange. Well I do believe its firmware issue so if u don't mind I'll mark it as firmware issue and close it. Though sometimes I'm thinking about logic that these issues shouldn't be closed, but then again its not up to me to fix them. I can't fix them.

outlyer-net, 2016-05-30:
I'm not familiar enough with the code to say wether it's a low level issue or an interface issue, so use your best judgement.

Jackysi, 2016-06-09:
This needs more investigation, at this time I don't use QOS so its extremely hard for me to test. If I enabled it temporary the graphs work.

Cu8eR, 2016-08-02:
Guys, test out the new v138 - it brings many fixes for QoS :)

closed 2016-10-02 by Jackysi