Jackysi/advancedtomato #169: 80 Mhz Channel width on Asus RT-AC68U

saintger, 2016-05-24:
Hello,

I've flashed tomato-RT-AC68U-AT-ARM-3.1-136-AIO-64K.trx on my Asus RT-AC68U and would like to use 80 Mhz as channel width on my 5 Ghz wifi.
I've selected Singapore as country in the Advanced Settings|Wireless|Wireless Settings (5 GHz / eth2) |Country/Region menu, and then selected 80 Mhz in the Basic Settings|Network|Wireless (5 GHz / eth2)|Channel Width.

However in the Status Overview Panel I can see that the Channel Width is always 40 Mhz for the 5 Ghz wifi.

Is that a bug in advancedtomato or in tomato ?

Thanks

saintger, 2016-05-27:
Workaround found by manually setting the 5 Ghz channel to 36 (instead of Auto). Then the channel width is indeed set to 80 Mhz.

Jackysi, 2016-05-29:
This is a tomato bug. AdvancedTomato is only GUI cover over Tomato by Shibby. Wireless issues are 99% related to Tomato.

closed 2016-05-29 by Jackysi