Jackysi/advancedtomato #170: PPPoE Connection issue

alecsandes, 2016-05-29:
Hi,

I've flashed to the 3.1-136 version and I cannot connect to the PPPoE connection.
I was trying to set the router from scratch so I don't upload any configuration.

Is it related to VLAN?
If so, can someone please help me to set this as I am not interested to set multiple VLANs and WANs

Thanks

alecsandes, 2016-05-29:
I tried to downgrade to 134 and I could not set the internet connection. All the tabs for Multi WAN were present in GUI but not accessible. The internet connection section was in non-accessible tabs.

Sorry, didn't made any screenshots. 

Went back to 136 and I could connect to PPPoE but actually I didn't have access to the Internet,
P.S. I made nvram clear after each action.

Jackysi, 2016-05-29:
You should really consider reporting these issues to Shibby. Does really noone read description what AdvancedTomato is? Its only GUI for Tomato by Shibby...

closed 2016-05-29 by Jackysi