Jackysi/advancedtomato #172: IP Traffic Real-Time and 24h views are blank with Asus RT-AC68U

saintger, 2016-05-30:
Router: Asus RT-AC68U
Advancedtomato: tomato-RT-AC68U-AT-ARM-3.1-136-AIO-64K.trx

Screenshots:
[image]
[image]
[image]
[image]
[image]
[image]

I haven't managed to display the IP traffic with Advancedtomato (either real-time or last 24h). The X-axis and Y-axis are updated (i.e. scaled according to the current IP traffic) but I don't see any plot (see screenshots).

Bandwidth monitoring works fine (I can see the plot), and other views (View Graphs, Transfer Rates, Daily and Monthly) seem to work fine.

IP Traffic monitoring is enabled in the Administration Panel (I haven't changed the default configuration) and I have tried with both static and dynamic IP. With static IP, I have checked "Bound To" and "IPTraffic" for each static IP (I have also tried without theses checkboxes checked).

I have search this issue and it seems that nobody else had this issue before. Perhaps I have forgot to activate an option somewhere ?

Jackysi, 2016-05-30:
Mm I honestly don't know for sure what's the problem (I'm only GUI developer and not exactly expert on Networks and Tomato). However I can say that its not "AdvancedTomato" issue it self (This is GUI on top of Tomato By Shibby) so in this case I have to close the issue.

If it were me, I'd seek help at http://linksysinfo.org/ and try again by defaults (resetting nvram and reflashing image). Also maybe check if there are javascript errors in F12 Console.

closed 2016-05-30 by Jackysi

saintger, 2016-05-30:
Just for info, I have installed Tomato by Shibby and the Real-Time IP Traffic monitoring works perfectly.
So I am not sure that the problem is in Tomato by Shibby.

Jackysi, 2016-05-30:
I have R7000 and RT-N66U and I can verify that Real Time IP Traffic works fine on both. With that in mind there is no custom image for AC68U and so I believe the issue is related to your router or NVRAM settings.