Jackysi/advancedtomato #173: r6300v2 on V1.0.4.2_10.0.74 - no WAN IP, DNS issues

zorrobyte, 2016-05-30:
I suddenly had no internet access last night, my modem appeared to be working just fine. I was running V1.0.4.2_10.0.74 for a few months now with no issues. In fact I've been running Advanced Tomato for the last year or two.

However no public IP address would be assigned to the R6300v2's WAN port (from a SB6131 modem on Comcast). I tried several power cycles, MAC address changes and finally a reflash and reset to factory Tomato settings. I let it sit all night thinking that it was a Comcast issue.

This morning I connected my modem directly to my PC and all was fine, was getting an IP as expected. Connected back to the R6300v2 and power cycled - nothing. I flashed back to stock firmware and it started working with no issues.

Another funky thing is that while the router was set to use GoogleDNS, I constantly had to do a 
```
sudo bash -c "networksetup -setdnsservers Wi-Fi Empty; networksetup -setdnsservers Wi-Fi 8.8.8.8; networksetup -setdnsservers Wi-Fi 8.8.4.4"
```
when connecting to my work VPN to access internal resources. On stock firmware I do not have to do this.

Best of luck with the project!

Jackysi, 2016-05-30:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato and so its not actually a firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-05-30 by Jackysi

zorrobyte, 2016-05-30:
Much appreciated! Reported.