Jackysi/advancedtomato #174: Bandwidth Limiter not working

AdziePahlevi, 2016-05-31:
Hi, recently I'm doing some experiment and implementing bandwidth limitations on some devices..
as you can see below the bandwidth limiter is enabled and I already bound one device with the IP, in my case is 192.168.100.43

[image]

however after its enabled and also after router is rebooted, seems the speed of the limited devices is still getting unlimited bandwidth, below is the from the IP Bandwidth monitoring

[image]

not sure why its not working, I already put devices MAC address and even I put the IP Address as well, but seems have no impact. would you please have look on it.

AdziePahlevi, 2016-05-31:
I found these on linksysinfo.org

http://www.linksysinfo.org/index.php?threads/trouble-with-bandwidth-limiter.72417/

could be related, some users are posting using 136 release the bandwidth limiter is not working, while it works on 124 release..

Jackysi, 2016-06-09:
Yeah, its defo Tomato issue and has nothing to do with AdvancedTomato GUI.

closed 2016-06-09 by Jackysi

Bezalel79, 2016-07-10:
Hi, Do you know if it has been fixed in version 137?
I've tried the 132 version but don't work
Thanks

AdziePahlevi, 2016-07-10:
on 132 bandwidth limiter is working like a charm, however in 137 after I flash it's not working.. people on linksysforum said I need to do clean flash router and re-configured the router from the scratch which is taking a lot of time, so I still haven't done it... if you want to test on 137 pls do it by performing clear flash your router...

Bezalel79, 2016-07-11:
I did some tests and I realized that in the original version 132 of the BWM Shibby works well but in the advancedtomato 132 version does not work .
Regarding the version 137 does not work or the Shibby version nor that of AdvancedTomato

AdziePahlevi, 2016-07-11:
I can confirm it works on 132 Advance tomato. Have you performing clean firmware installation using ASUS Firmware restorations?

Bezalel79, 2016-07-11:
I have not been able to do clean firmware installation using ASUS Firmware restorations with the 132 version of advanced tomato becouse I received an error and my Aus rt-n16 not go out from the Recovery Mode. Using Shibby version it's works.
In my test I start from Shibby version, migration to advancedtomato,  clean nvram no restore configuration

AdziePahlevi, 2016-07-11:
I discover this issue when I'm using advance tomato 136 then I create this thread. Later I found this problem also observed on Shibby release. Last time I downgrade from 136 to 132 using Asus Firmware restorations.. 

try 30/30/30 method to put your router to Maintenance Mode, its also difficult for me last time but it was goes to maintenance mode after 30/30/30

Below is the steps:
1. Press Reset button (small black button) for 30 seconds, after 30 seconds plug out the power
2. Keep press the reset button for another 30 seconds, after 30 seconds passed, plug in the power
3. Keep Press reset button for another 30 seconds.

voila it goes to maintenance mode.

Bezalel79, 2016-07-12:
Forgive me for my bad English . My problem is just the opposite . If I try to upgrade to the Advanced version Tomato 132 router does not go out from the recovery mode. Power led blinking slowly after reboot.