Jackysi/advancedtomato #180: Tomato Firmware Causes Bootloop for Asus RT-AC68U Hardware Rev C1

buddwm, 2016-06-20:
Appears that new hardware revision of RT-AC68U is not compatible with latest Tomato firmware.  The C1 revision appears to have a processor upgrade which may be the reason for the incompatibility.

buddwm, 2016-06-23:
Posted this in wrong section, so closing.

closed 2016-06-23 by buddwm

fonix232, 2017-04-12:
Has there been any progress on this?

I have an Asus RT-AC66U_B1 (no, the B1 is not the HW. Rev., it's the name of the router, confusing, right?), which is practically an RT-AC68U C1 in a horizontal case. Same basic HW, same features, apart from N being limited to 450Mbps from the AC68U's 600Mbps.

The stock firmware is buggy, DD-WRT is crashing all the time, and I can't get Merlin to work as intended (the DHCP forwarding part does not work).

I'd lovely test (and possibly compile) the firmware, if there was some preliminary support added.