Jackysi/advancedtomato #184: Access Restriction Port/Application ip address

alphading89, 2016-06-27:
Inside the Access Restriction, the rules cannot be more than 20 entries (shown in red square in the screenshot).
I like to block 199.16.156.38, 199.16.156.70 and so on.
is it possible to do a range ip address like, "199.16.156.0 - 199.16.156.254"? or any other way that can do a range ip address in the Rules section?
Please advise.
Thank you.
[image]

Jackysi, 2016-07-10:
Not sure but I think this is limitation set by Firmware it self so I might not be able to do anything about it. I will give it a look though

alphading89, 2016-07-11:
thanks, please update me if you figure something out..

Jackysi, 2017-11-14:
Sorry I'm getting back sooo late. I've been looking into this and found out that it may be possible to set ranges. However, my knowledge is pretty limited to the way this works in back-end so I can't extend functionality this far. 

Maybe some other developer with enough C knowledge will be able to figure it out.