Jackysi/advancedtomato #187: Wireless survey (under Tools) inoperative

grimley, 2016-07-09:
Kudos on the work.

Jackysi, 2016-07-10:
I will need more details, otherwise I will close this issue. Like screen shot of what is wrong? My router has no problem on that page.

grimley, 2016-07-10:
Nothing appears on mine. 
http://192.168.1.1/#tools-survey.asp 

[image]

Jackysi, 2016-07-10:
Did u click refresh? It doesn't appear unless refresh is setup... That's normal

grimley, 2016-07-10:
That did it. Auto Refresh is on and I refreshed the page. Did everything but hit Rerfresh. 
Thanks

Jackysi, 2016-07-10:
^^ ur welcome!

Cu8eR, 2016-07-31:
Maybe add a note below or make a button in a different color would solve this issue

closed 2016-07-31 by Jackysi