Jackysi/advancedtomato #188: Asus AC-68 with verizon 4G USB modem 551L issue connecting to certain websites/ips

jwprice, 2016-07-16:
I'm using Tomato Firmware 1.28.0000 -3.2-137 K26ARM USB AIO-64K on a ASUS AC-68 router. 
I connect to the internet via a 4g USB Verizon 551L Modem. 
Connectivity was easy to setup but I am having a major problem with amazon.com for some strange reason.
Firesticks via the router seem to be unaffected by this problem to make you even more puzzled... 

The router can't access it directly via the 4g modem... My laptop directly connected to the USB modem can access amazon fine.
But when it's attached to the router; neither laptop, phone, nor even the ping tools via the Advanced Tomato GUI can get a ping or connect. 

If I use Chrome with the HOLA vpn widget on and using a same country (USA) person to proxy through, I can connect fine through the router.. Which is fine for ordering, books, or prime music, but not for video as it buffers too much. 

Can you help me please? It seems it's a problem via the router and not the card itself as it is fine when connecting to any other site or using my laptop directly with the card for amazon.

closed 2016-07-17 by Jackysi

jwprice, 2016-07-18:
Why was my problem closed? I already explained the problem in pretty lengthy detail that it is a issue within the firmware. It does not happen on merlin or ASUS's as I've tested with my second ASUS RT-AC68 model at home.

Jackysi, 2016-07-18:
There is 166 closed tickets out of which 150+ are false reports. AdvancedTomato is a GUI modification for Tomato and so its not actually a firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

jwprice, 2016-07-21:
Thanks for the info.