Jackysi/advancedtomato #189: Add support for AP Isolation for Virtual Wireless

KazWolfe, 2016-07-17:
It is currently impossible to enable AP Isolation for virtual wireless networks (like `wl0.1`) though the Advanced Tomato Control Panel.

This seems problematic as many people depend on this feature to protect and secure networks where users should be unable to access other resources (for example, a guest wifi). Would it be possible to get a check box for this?

Cu8eR, 2016-07-31:
Is it in shibby fw or completely a new feature?

KazWolfe, 2016-07-31:
I'm not sure if it's present in Shibby as I've never used it, nor do I have a device for it, but I know that it's present in DD-WRT.

nobodypb, 2016-08-01:
I don't have a running advanced tomato here, but according to
https://github.com/Jackysi/advancedtomato/blob/advancedtomato-ac/release/src/router/www/advanced-wireless.asp#L107
There should be AP Isolation on the Advanced Wifi page

burntoc, 2017-01-16:
It is an option is DD-WRT, but in Advanced Tomato, you can only enable/disable it at the physical wireless adapter level.  You can't specify virtual APs to have it enabled while leaving it off on others as you can in DD-WRT.

Jackysi, 2017-11-14:
These functions are there only for actual wireless interfaces (Advanced Settings -> Wireless) but not for Virtual Wireless Interfaces. So I will call this closed, because I don't think it will ever get to Virtual Interfaces

closed 2017-11-14 by Jackysi