Jackysi/advancedtomato #192: Add support for Bandwidth Limiting for other LANs

KazWolfe, 2016-07-17:
Currently, the AdvancedTomato Control Panel only allows admins to set bandwidth limits for br0.

Other bridges (like br1) allow admins to choose a class for traffic, but it doesn't allow customized limits for IP ranges and similar.

Cu8eR, 2016-07-31:
Is it something what is missing? I am not sure if I understand this issue correctly - but in shibby fw v138 I can set this to every br interface
[image]

KazWolfe, 2016-07-31:
This feature request is not for the "Default Class" interface. It's for the actual "Bandwidth Limiter" feature, which only exists for `br0`.

[image]

Either this should get added, or we should be able to make each specific device have a bandwidth limit, instead of the universal "the bandwidth will be shared by all hosts" mode that it's currently using.