Jackysi/advancedtomato #195: Multiwan - Failover for no reason and Failback is extremely long

Modjor, 2016-07-23:
Hi,

1st thing first : thanks for the work around this, it's an extremely pleasant CFW to work with.

But now, the pain... ;)

-------------------------------------------------------
**ENVIRONMENT:**

- Asus AC68U
- AdvabcedTomato V3.2-137
- MultiWAN configured with 2 WAN
- WAN 1 is linked to the Asus WAN port, - connected to a D Link DSL Router
- WAN 2 is linked to the Port #1 of the Asus, uses DHCP - connected to a Raspberry Pi acting as router with a 4G dongle
- "Check Connection every" set to 2 minutes - Target 1 is www.google.fr / target 2 is 8.8.8.8

WAN 1 SETUP:
- Asus WAN Port connected to a DSL Router
- Get a private IP assigne from the DHCP Server of the DSL Router - DHCP lease is 48 Hours
- Load balancing Weight is 1

WAN2 SETUP:
- WAN2 is linked to the Asus ethernet port #1, connected to a Raspberry Pi acting as 4G router
- Get a private IP assined by a DHCP server hosted on the Raspberry (DHCP lease is 7 days)
- Load balancing weight is 254

Goal of the above setup:
- I have a very slow DSL line (1Mb down 300KB Up), and a very good 4G connection bu capped at 50Gb per month.
- I've setup a static MultiWANrouting rule so my NAS is always downloading torrent and stuff using the DSL line and NEVER go to the WAN2 4G connection (so it does not consume my entire 4G data caps). Simple rule based ont the source IP address.
- All other traffic (websurf, video streaming etc..) is normally going through the 4G WAN connection
- If WAN2 become unavailable, all traffic is routed through the WAN1 DSL connection

NOTES / CONTEXT:
I originally configured WAN1 with a "Load Balancing Weight = 0", to use it as failover for WAN2, but it seems that in this mode WAN1 become active only when WAN2 is failing. The resulst of that was that my NAS was unable to use WAN1 as long WAN2 was OK.
That's why I change the WAN1 Weight to 1 and WAN2 weight to 254. According to my understanding, the traffic should be primarly routed trhought WAN2, and if WAN2 become unavailable/too slow traffic should be rerouted through WAN1. At least this is what I understand, and seems to work most of the time.

-------------------------------------------------------

**ISSUE DESCRIPTION:**

Problem #1 - Unexpected Failover
Once and a while (seems random), the router detect a failure on WAN2 and reroute all traffic to WAN1.
However, I can confirm WAN2 is not failing - if I connect my laptop directely on the Raspberry Pi subnet I got Internet connectivity.
The 4G connection is rock solid, 15Mb Down, 3Mb up anytime of the day / any weather etc...
Also, the network link between the Asus and the Raspberry is OK: I can ssh from the Asus LAN to the Raspberry
But well - Let's admit that once and a while, the 4G network become unavailable for a very brief period of time, and that may explain why sometime the router pick up on a link unavailable and start a failover.

Problem #2 - Extremely long Fallback
I can imagine there is maybe there is a very short network unavailability from the 4G network, causing the Advanced Tomato to execute a Failover to WAN1. 
But the other problem is that after that, it can take a very long time to fallback to WAN2.
 If you look at the log below, you can see that a failover seems to happen at 7.34AM, but the failback only occurs at 11.47AM. I can confirm that during this period of time, traffic as routed through my DSL line (WAN1), but the 4G connection as perfectly up and running (test done using my laptop behind the Asus and then behind the 4G Raspberry)

---------------------------
LOG

Jul 23 07:28:20 asus daemon.info dnsmasq-dhcp[31108]: DHCPREQUEST(br0) 192.168.5.93 20:0c:c8:19:7e:fc 
Jul 23 07:28:20 asus daemon.info dnsmasq-dhcp[31108]: DHCPACK(br0) 192.168.5.93 20:0c:c8:19:7e:fc DGN2200v4
Jul 23 07:34:00 asus user.warn mwanroute[7901]: Multiwan status is changed, last_status=0000, now_status=2200, Update multiwan policy.
Jul 23 07:34:00 asus user.emerg mwanroute[7901]: OUT fun mwan_load_balance, mwan_curr=2200
Jul 23 07:50:50 asus daemon.info dnsmasq-dhcp[31108]: DHCPREQUEST(br0) 192.168.5.93 20:0c:c8:19:7e:fc 
...
...
Jul 23 11:43:53 asus kern.warn kernel: br0: received packet on eth1 with own address as source address
Jul 23 11:45:34 asus kern.warn kernel: br0: received packet on eth2 with own address as source address
Jul 23 11:47:00 asus user.warn mwanroute[28635]: Multiwan status is changed, last_status=0000, now_status=2200, Update multiwan policy.
Jul 23 11:47:00 asus user.emerg mwanroute[28635]: OUT fun mwan_load_balance, mwan_curr=2200
Jul 23 11:47:27 asus kern.warn kernel: br0: received packet on eth1 with own address as source address
Jul 23 11:48:25 asus kern.warn kernel: br0: received packet on eth1 with own address as source address
Jul 23 11:49:22 asus kern.warn kernel: br0: received packet on eth2 with own address as source address
---------------------------

Any help would be extremely appreciated - and I currently have a lot of free time if you need some testing to be done - so feel free to ask me anything you need to troubleshoot this one, I'd be glad to help.

Jackysi, 2016-07-26:
Oh, I hope I would be able to do something about this but I can't. I'm not actual firmware developer I only work on the GUI =/. Would you mind copying this issue to https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open and I'm sure Shibby will do something about it. Advanced Tomato is a GUI modification based on Tomato by Shibby, its not really a separate firmware but it does need some modifications other than GUI so that's why I compile my own images.

closed 2016-07-26 by Jackysi

Modjor, 2016-07-26:
I'm sorry - I did not catch that! Will definitely open a ticket on Shibby bug tracker.
Anyway, your gui is actually very nice, keep up the good work :)

Jackysi, 2016-07-26:
Thanks =) I will!