Jackysi/advancedtomato #201: Enable/disable multiple access restriction rules at once

aln-1, 2016-08-01:
It could be nice if the access restriction "enable" checkbox will be on the overview list of access restrictions together with a "save" button.

In this way you could enable/disable multiple access restriction rules at once.

This is done the same way in asuswrt.

Jackysi, 2016-08-01:
Most likely I can't do anything about this because the check boxes are handled differently in the back end than normal form submission, but I will give it a look.

Jackysi, 2017-11-14:
I've carefully reviewed this enhancement request and I've came to conclusion that this is not possible unless  certain Tomato back-end handlers are changed. Since I'm trying to keep this project GUI modification only I can not add this (question is if I even could due to my limited C knowledge). However, this is good feature and would be good to have it, so I will leave this open and maybe one of other developers will add it in future to the Tomato Firmware it self.