Jackysi/advancedtomato #206: VLAN Settings Blank in Version 3.3-138

ike200000, 2016-08-14:
ASUS RT-AC66U
Version 3.3-138

Just upgraded from 137 to 138 and the VLAN settings are blank. I restored default settings and cleared NVRAM. This did not fix anything.

[image]

Jackysi, 2016-08-14:
Can you open up F12 Console and try that again?

ike200000, 2016-08-14:
[image]

Jackysi, 2016-08-14:
Hmm that's hard to debug without device. Sorry, I can't fix it like that.

ike200000, 2016-08-14:
Ok so the first error when loading the page is on the jquery.min.js script. "SyntaxError: Unexpected Token }"
[image]

Digging further shows that the "port_vlan_supported" is undefined. This is strange since it is defined in the NVRAM.
[image]

[image]
...
[image]

Continuing on shows that trunk_vlan_supported is also not defined multiple times.
[image]
[image]

Then another "port_vlan_supported" error.
[image]

Next is a tomato.js null error on the style property
[image]

BrainDedd, 2016-08-15:
Just want to add my +1 on this. ASUS RT-N66U.

Jackysi, 2016-08-15:
I have N66U + latest firmware and I don't see this problem... How is that possible?

pdonadeo, 2016-08-16:
Same problem with 	Asus RT-N16

[image]

pdonadeo, 2016-08-16:
If I can help in any way, tell me what you need to debug this situation.

Jackysi, 2016-08-16:
Fixed: https://github.com/Jackysi/advancedtomato-gui/commit/1bda0c3e0abdab9fdf41f72fbd68cb73f475a0ec

BrainDedd, 2016-08-16:
Hmmm ... we've all forgotten to close a comment block :smile: 
Look forward to grabbing the new version :+1:

Jackysi, 2016-08-16:
I've uploaded new images with the above fix. The naming is the same however the MD5 checksum and the Images are new. Please try and report.

BrainDedd, 2016-08-17:
Working :+1:

bgt59, 2016-10-10:
Hi,
It seems this issue is still present for WNR3500Lv2 too.

itsishrak, 2016-10-10:
Same issue on the N10U. 
[image]

nsenica, 2016-10-14:
I can confirm that this is still happening on a N10U, the trx was downloaded yesterday from the main website.

ChrisHughes, 2016-10-17:
Still happening on an RT-N53

Jackysi, 2016-10-18:
can someone post whole page source (not by using view source, but using F12 -> click on <html> -> edit html and copy all and paste here using "INSERT CODE" - Chrome, Idk for other browsers). I will give it a look but i'm not sure what is missing.

nsenica, 2016-10-18:
Here it goes.

I think I can point you directly to the problem, just search for 
```
<input type="hidden" name="lan_ifnames">
```
And you'll see that this starts in the middle of the vlg.dataToView JS declaration.

Let me know if you need more information

Thanks
Nuno

```
<html lang="en"><head>
<meta charset="utf-8">
<meta name="viewport" content="width=device-width, initial-scale=1.0">
<meta http-equiv="content-type" content="text/html; charset=utf-8">
<meta name="robots" content="noindex,nofollow">
<title>[TomatoUSB] VLAN</title>

<link href="css/interface.css" rel="stylesheet">

<link rel="shortcut icon" href="favicon.ico">

<script type="text/javascript" src="js/jquery.min.js"></script>
<script type="text/javascript" src="tomato.js"></script>
<script type="text/javascript" src="js/advancedtomato.js"></script>

<script type="text/javascript">
var wl_ifaces = {};
var routerName = '[TomatoUSB] ';
//
nvram = {
	'at_nav': '',
	'at_nav_action': 'click',
	'at_nav_state': 'default',
	'at_update': '',
	'tomatoanon_answer': '1',
	'wl_unit': '0',
	'http_id': 'TIDdbec541aa737c6e2',
	'web_mx': '',
	'web_pb': ''};

//
anonupdate = {
};

// AdvancedTomato related object
var gui = {
'ajax_state'   : false,
'nav_delay'    : null,
'nav_action'   : ( ( typeof(nvram.at_nav_action) != 'undefined' && nvram.at_nav_action == 'hover' ) ? 'mouseover' : 'click' ),
'refresh_timer': null,
'version'      : "1.28.0000 MIPSR2-3.3-138 K26 Max",
};
// On DOM Ready, parse GUI version and create navigation
$( document ).ready( function() {
// Attempt match
match_regex = gui.version.match( /^1\.28\.0000.*?([0-9]{1,3}\.[0-9]{1}\-[0-9]{3}).* ([a-z0-9\-]+)$/i );
// Check matches
if ( match_regex == null || match_regex[ 1 ] == null ) {
gui.version = 'More Info'
} else {
gui.version = 'v' + match_regex[ 1 ] + ' ' + match_regex[ 2 ];
}
// Write version & initiate GUI functions & binds
$( '#gui-version' ).html( '<i class="icon-info-alt"></i> <span class="nav-collapse-hide">' + gui.version + '</span>' );
AdvancedTomato();
});
[... 751 lines cut ...]
<br>
</div>
</div>

<script type="text/javascript">
if(!port_vlan_supported)
$('#sesdiv').after('<i>This feature is not supported on this router.</i>');
else {
E('sesdiv').style.display = '';
if(!trunk_vlan_supported)
E('trunk_vlan_override').style.display = '';
}
</script>
<button type="button" value="Save" id="save-button" onclick="save()" class="btn btn-primary">Save <i class="icon-check"></i></button>
<button type="button" value="Cancel" id="cancel-button" onclick="javascript:reloadPage();" class="btn">Cancel <i class="icon-cancel"></i></button>
<span id="footer-msg" class="alert alert-warning" style="visibility: hidden;"></span>

<script type="text/javascript">earlyInit(); verifyFields(null,1);</script>
</div>
<div class="clearfix"></div>
</div>
</div>

</body></html>
```

nsenica, 2016-10-18:
I've also checked inside the router, and the advanced-vlan.asp file is exactly has only 653 lines, and it looks like this: 

```
(data[ COL_P3 ].toString() != '0') ? 'Yes' : '',
(data[ COL_P3T ].toString() != '0') ? 'On' : '',
(data[ COL_P4 ].toString() != '0') ? 'Yes' : '',
(data[ COL_P4T ].toString() != '0') ? 'On' : '',
(data[ COL_VID_DEF ].toString() != '0') ? '*' : '',
[ '', 'WAN', 'LAN (br0)', 'LAN1 (br1)', 'LAN2 (br2)', 'LAN3 (br3)', 'WAN2'
<input type="hidden" name="lan_ifnames">
<input type="hidden" name="lan1_ifnames">
<input type="hidden" name="lan2_ifnames">
<input type="hidden" name="lan3_ifnames">
<input type="hidden" name="vlan0vid">
<input type="hidden" name="vlan1vid">
<input type="hidden" name="vlan2vid">
```

Which confirms the problem. My guess, if I may, is that the firmware image on the website is not updated. 

Regards,
Nuno

Jackysi, 2016-10-19:
Thanks! I will compare with my version of the file and then see if the problem is in both or just the version from the web site. I do believe too that I forgot to recompile ALL firmware's because I thought the issue is only part of ARM. Not sure.

I have some updates so I'll release 3.4-138 with this fix soon.

nsenica, 2016-10-19:
Awesome @Jackysi :) Let me know if you need further investigation from my side.

Jackysi, 2016-10-19:
I could use a tester, I just need 1 router name to compile fast 1 image. I think I've found out the issue, it was huge :D

nsenica, 2016-10-19:
I can test it, no problem :) Mine is a RT-N10U, if you want to compile the image I can test it later on.

Jackysi, 2016-10-19:
MD5: c46ceee44cace92ff9145a67c3bca34c 
https://advancedtomato.com/data/tomato-K26-1.28.AT-RT-N5x-MIPSR2-3.3-138-Max.trx

I think that should be the one

nsenica, 2016-10-19:
Hi @Jackysi 

I'm really sorry to tell you, but that is broken as well :( Upgrade via UI didn't work, I could ping the router but I couldn't access it via http or ssh.

I needed to do a regular flash, uploaded the firmware via Asus Firmware Restoration tool, and I got the same thing again :(

Any idea ?

Jackysi, 2016-10-19:
So this image failed completely? Ahm that's rather strange

nsenica, 2016-10-19:
I'm not sure what you mean with "failed completely", but I suppose that after the firmware restoration it loaded correctly. 

Can you provide a image with some easily identifiable change so we are sure if it was loaded correctly or not ?

Also, it seems that the asp files are somehow preprocessed, could it be that the pre-processor doesn't like the comments within the JS section ? Can you remove them and see if it works ?

Thanks

Jackysi, 2016-10-19:
The problem is not preprocessor (which actually must be there) so the firmware fits certain routers. The problem is non-closed comment in the javascript which has been fixed in this commit: https://github.com/Jackysi/advancedtomato-gui/commit/1bda0c3e0abdab9fdf41f72fbd68cb73f475a0ec

Now the only problem is images on the web site don't include that fix.

nsenica, 2016-10-20:
Ok. I can help you testing it, just let me know when.

:)

nsenica, 2016-10-21:
@Jackysi , I noticed something that I'm not sure if it doesn't break somewhere in the build process.

There are /* MULTIWAN-BEGIN */ and /* MULTIWAN-END */ in the middle of HTML, in HTML /* and */ don't have any meaning as comment delimiter, could the pre processor be breaking the build due to interpreting that as regular comments ?

Just wondering.

Jackysi, 2016-10-21:
I have already given you fix for the VLAN page error. I don't really understand what are you talking about further ^^

nsenica, 2016-10-21:
hmm I'm not sure I understand what you say, but the image you provided in this [comment](https://github.com/Jackysi/advancedtomato/issues/206#issuecomment-254802985) didn't work, as I stated before. So I assumed that the fix was not working.

Jackysi, 2016-10-21:
You said the image I attached  didn't flash properly so it broke your router no? How do you assume that VLAN page wasn't fixed? If  VLAN page had error it wouldn't brick whole router cause of it. I'm not sure why your router would brick, but I flashed the image on my Asus RT-N66U and the VLAN page works good.

Jackysi, 2016-10-21:
P.S.: Whats the name of image you usually flash on your N10? I have some more images but idk which to give you

nsenica, 2016-10-21:
The image didn't upgrade (via UI) properly at first, but then I managed to flash it using the Asus Firmware Restoration app, and after that the VLANs page is/was still broken. 
Maybe the flash didn't work and it loaded what I had there initially. I can try to flash a previous version (127) first, and then try to upgrade to this one you provided again.

I usually flash from http://tomato.groov.pl/ and for this model K26RT-N, this was the first time I tried flashing your firmware.

HFsi, 2016-10-22:
Exactly same problem for a Linksys E2500 V3

barkster, 2016-11-02:
Is this fixed for RT-N16, I reflashed with latest download and mine are still blank

Jackysi, 2016-11-02:
the fix will come with 3.4-138 or 3.4-139

closed 2016-11-02 by Jackysi

charsi, 2016-11-18:
I am seeing this issue in WNR3500Lv2 - 3.3-138. Is there anything I can change without having to build an image from scratch to get this working for the time being? 
Happy to do some testing too.