Jackysi/advancedtomato #209: Too big image

maciejby, 2016-08-14:
Latest build of firmware 3.3-138 for Tenda N60 is too big. Can't upgrade.

Jackysi, 2016-08-14:
Ill try to remove some feature, though that means there will be a feature less =/

upgap, 2016-08-25:
Same problem here. Firmware 3.3-138 for Tenda N60 is too big. Why isn't it removed from download ? (until the issues are resolved) What's the point of providing a firmware no one can flash ?   Anyway, I've bought a new (supported) router today with more than 8 MB Flash RAM to get rid of  those problems hopefully for very long time.

Jackysi, 2016-08-25:
Some people were successful flashing it so I'm not removing it for those who are. Don't ask me how that is possible.

closed 2016-12-03 by Jackysi