Jackysi/advancedtomato #214: Unable to save in the 'USB support' section

thyeestes, 2016-08-16:
Router: Netgear Nighthawk R7000
Firmware: Tomato Firmware 1.28.0000 -3.3-138 K26ARM USB AIO-64K

I try to change settings in the 'USB support' section, but I'm unable to save. The cancel button does work and reverts the settings. I tried to erase all data in NVRAM, but after that the save button still does not work. Anyone else with the same problem?

Jackysi, 2016-08-16:
https://github.com/Jackysi/advancedtomato/issues/210

closed 2016-08-16 by Jackysi

thyeestes, 2016-08-16:
Sorry did not see that one. Although I do not understand how to fix it

Jackysi, 2016-08-16:
You can't fix it. If you want the fix you need to compile new image with that change in it. Unless you know how to hack read-only file system on the router and push the specific file to the router.

thyeestes, 2016-08-16:
No that is way over my head :-) I'll wait for the next release.

Jackysi, 2016-08-16:
Actually I'm going to recompile images soon, because there is also issue with VLAN however it will be the same version so might be little confusing, will let u know when uploaded.

thyeestes, 2016-08-16:
Yes please.

thyeestes, 2016-08-16:
Don't know if it is a bug, but on the 5GHz band I can not select the channel, only auto.

Jackysi, 2016-08-16:
Its done, uploaded. Not a bug change location (Singapore gives the most options)