Jackysi/advancedtomato #217: Captive portal stops working after a few hours.

BowenLibrary, 2016-08-19:
We are trying to use the captive portal function at our very small public library but after a few hours it will no longer allow new users to connect.

Previously connected users can continue to work but new ones can not connect to the WiFi network.

We're running AdvancedTomato v3.3-138 on a brand new Netgear R7000 but we had the exact same problem with our ASUS RT-N66U.

Any suggestions of settings to try would be greatly appreciated.  I am not an experienced network person so I'm sure there are settings which I need to change which would be obvious to a more experienced person.

I'm using mostly default settings in AdvancedTomato, but I've got a very simple custom splash page loaded into the captive portal settings and it's working perfectly.

But as I said, after a few hours new users can no longer connect to the network at all, never mind reach the splash page.

Jackysi, 2016-08-28:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato by Shibby and so its not an actual firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-08-28 by Jackysi

BowenLibrary, 2016-09-06:
Thanks very much for the explanation and the link Jackysi, it is much appreciated!

I am following up with Shibby.

Leo