Jackysi/advancedtomato #221: QOS scheduler problem

szwistak, 2016-08-26:
Hi,

I'm  trying to enable QOS on Ac66u and Im getting "The field "qos_pfifo" is invalid. Please report this problem." I didnt touch any of setting bellow. Just changed scheduler to fq_codel. 

thanks,
Rafal

Jackysi, 2016-08-28:
Will be looked into

maciejby, 2016-09-02:
Same here. Tenda N60

Jackysi, 2016-10-02:
The scheduler fq_codel is only available on ARM. The GUI shows it because sometimes I miss changes that shibby does between branches (RT-AC/ARM/RT-N). Anyway, fixed.

closed 2016-10-02 by Jackysi

Jerezano, 2016-10-05:
I still have this issue on E2500v1

f00b4r0, 2016-11-11:
Issue still exists in v3.3-138 on E2500v3. codel and fq_codel options are invalid as they are not supported by the provided kernel.

Jackysi, 2016-11-12:
The fix applies to version 3.4-138+.