Jackysi/advancedtomato #223: L2TP in 3.3-138 on RT-N16 has a bug that isn't present in Shibby's v138

eierfrucht, 2016-09-05:
To be more specific, L2TP connection status is always shown as DISCONNECTED -- even though L2TP is working and I'm able to surf and download --  if I use https://advancedtomato.com/get/v3.3-138/K26RT-N/Asus%20RT-Nxx/tomato-K26USB-1.28.AT-RT-N5x-MIPSR2-3.3-138-AIO.trx on my RT-N16 but everything is okay if I flash Shibby's v138.

Both AdvancedTomato 3.3-138 and Shibby's v138 work without issues on my other router, RT-N18U.

Here's what I'm getting with AdvancedTomato 3.3-138 on my RT-N16:

[image]

Please notice that subnet mask is always shown as 0.0.0.0 in the status page no matter which value I input in the Basic Settings tab:

[image]

Also, the VLAN settings page is always blank and neither of the two buttons is responsive:

[image] 

Before a working L2TP connection is established (even though the status shown in the GUI is always DISCONNECTED), I get bunches of these messages in the log:

`Sep  5 07:40:54 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 07:41:02 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 07:41:02 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:41:09 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:41:09 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:41:17 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:41:17 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:41:25 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:41:25 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:41:33 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:41:33 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:41:42 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:41:42 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:41:51 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:41:51 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:41:59 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:41:59 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 63513, tunnel = 6569 Dumping.
Sep  5 09:42:07 unknown daemon.info xl2tpd[1658]: Can not find tunnel 6569 (refhim=0)
Sep  5 09:42:07 unknown daemon.debug xl2tpd[1658]: network_thread: unable to find call or tunnel to handle packet.  call = 0, tunnel = 6569 Dumping.`

And this is what I have in my /etc/xl2tpd.conf

`[global]
access control = no
port = 1701
debug avp = no
debug network = no
debug packet = no
debug state = no
debug tunnel = no

[lac wan]
lns = 10.255.255.254
tx bps = 100000000
pppoptfile = /tmp/ppp/wan_options
redial = yes
max redials = 32767
redial timeout = 600
tunnel rws = 8
ppp debug = no`

Again, this only affects the AdvancedTomato build 3.3-138 for RT-N16. Shibby's build is unaffected and both builds work for RT-N18U. And yes, of course, I tried a full NVRAM reset for many times because it does look like corrupted NVRAM; unfortunately it isn't -- even after a thorough NVRAM erase, the issues persist :(

Jackysi, 2016-09-06:
Did u also try re-downloading latest images? For VLAN page there was a fix. I'll check for other issue, when I have the time. There is no differences other than the GUI so I'm not sure why would there be a difference in error logging.

eierfrucht, 2016-09-06:
I flashed both my RT-N16 and my RT-N18U with AdvancedTomato on the same day. RT-N18U had the VLAN page while RT-N16 didn't have. I just redownloaded and reflashed the latest image, same situation again. L2TP does connect normally on RT-N18U, connects with lots of errors on RT-N16 like before -- "network_thread: unable to find call or tunnel to handle packet" and such like stuff. Connection status is either Connecting or Disconnected. However after a bunch of the above mentioned errors, I'm able to ping web hosts, surf, download, whatever. Shibby's AIO v138 builds work equally well both for RT-N18U and RT-N16. I suggest that the latest AIO build of AdvancedTomato for RT-N16 is in some way corrupted.

eierfrucht, 2016-09-08:
Sorry for deceiving you, confirmed this to be a TomatoUSB issue. After multiple attempts, I was able to recreate these issues with the standard TomatoUSB (by Shibby) builds. The service responsible for checking WAN connection status fails to track Connected/Disconnected and uptime, so the router cannot shut down an existing L2TP connection before initiating a new one. Impredictable L2TP behaviour as a result,  let alone the fact that the L2TP server does not expect more than one simultaneous L2TP connection at a time.

AdvancedTomato is much more prone to this, can't say exactly why. Maybe some scripts start faster, and the whole thing happens because of a race condition?

Nevertheless, I submitted this bug to Shibby's issue tracker, dunno if he checks it or not. RT-N18U is unaffected while RT-N16 is.

This issue is not present in builds prior to the introduction of the MultiWAN feature.

If you have Shibby's e-mail, please inform him of this issue.

Jackysi, 2016-09-12:
Its okay, most of reports are firmware related anyway. I wish I had the knowledge to fix these issues but I simply don't. So I'm just closing every thing which is rather bad too because other developers can't help. But Imagine the amount of reported issues if I didn't do what I do... Since I can't fix them it would just go on forever and then the actual GUI issues would never be fixed. Blah.

I don't have Shibby's email, in fact I hardly ever get any contact with him, hes a busy man (once in 3 months maybe).

closed 2016-09-12 by Jackysi