Jackysi/advancedtomato #231: 3G modem issue

Jim101, 2016-09-23:
I am currently using ASUS RT N18U router. Flashed the Tomato advanced firmware. I am having USB problems in connecting a USB Modem to this router. I used the basic setup/WAN and chose 3G from the drop down list. After which I went to USB support attached device section and tried to mount the device. The device is shown on the list. When I click on the mount device. It will not mount. It gives out this message “Failed to mount verify the device is plugged in and try again”. The basic settings section states status connecting. I have tried to reboot the device and also I have re-flashed the firmware but nothing seems to work. Why is this? What am I doing wrong?

Jackysi, 2016-09-23:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato by Shibby and so its not an actual firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-09-23 by Jackysi

Jim101, 2016-09-23:
RT-N18U-AT-ARM-3.3-138-AIO-64K.trx