Jackysi/advancedtomato #233: Build for RT-AC3200

Pezmc, 2016-09-25:
Tomato[ has support ](http://tomato.groov.pl/?page_id=69)for the RT-AC3200 by Asus but there isn't a download on your website yet. Is it possible for you (or me!) to build a copy of advanced tomato against that hardware?

Jackysi, 2016-09-25:
First off, this is not support section nor a correct way to contact me. 

The ARM7 is currently on hold because I can't test any of the images before releasing them since I don't own any device compatible with the chip set. There is also some specific changes to the ARM7 which makes such device requirement. So this is on hold indefinitely.

closed 2016-09-25 by Jackysi

Pezmc, 2016-09-25:
How would you prefer to get in touch? A Github issue seemed more appropriate than an email, is there another repo for the advanced tomato website?

I have a `RT-AC3200`, is there a way I can help you test the builds?

Jackysi, 2016-09-25:
Email is more appropriate when it comes to such requests. No you can't, I need physical machine to work on because its not just builds. There are GUI changes and VLAN changes as well so it goes trial and error images after images before a specific "branch" can be compiled. I actually don't like that but that's how Shibby and other devs made it so there it is.