Jackysi/advancedtomato #237: "Auto resolve addresses" is always reset on entry to Quality of Service/Transfer Rates

vrapp, 2016-10-08:
Whenever I select Quality of Service/View Details, or Transfer Rates, the IP's in Source and Destination are not resolved, even though the option to auto-resolve is permanently checked in Filters.

To reproduce:
Go to Quality of Service/Transfer Rates
expand Filters, and check "Auto resolve addresses"
The IP's will become resolved
go to any other node in the menu
return to Service/Transfer Rates; expand Filters again, and confirm that "Auto resolve addresses" is still checked.

Expectation: the IP's in source and destination should be auto-resolved at once
Reality: they are not; the cursor appears as waiting (spinning wheel), and it does not change, 

Uncheck and re-check "Auto resolve addresses" - the IP's will become resolved immediately, and spinning wheel will change to normal cursor.

[image]

The same in  Quality of Service/View Details.

Also, when the IP's are not resolved, click on the "resolve" shortcut has no effect.

Jackysi, 2016-10-08:
Does this work on Shibby?

vrapp, 2016-10-08:
> Does this work on Shibby?
No idea, I only have advanced tomato. If I understand correctly, Advanced is about web interface, so since this is purely web interface issue, it's probably AT-specific.

Jackysi, 2016-10-08:
That's not true. The data for the web interface comes from the firmware (custom C based HTTPD handler) which I don't have knowledge to fix. And so its possible that this issue is present on both firmwares. 

I will give it a look when I have time.

Jackysi, 2016-10-11:
I've checked this and i'm pretty sure this bug is Tomato related not just AdvancedTomato. I don't think it will be fixed anytime soon because there are more pressing issues to take care first. But I will have it as a note maybe to fix it one day.

Jackysi, 2017-11-14:
Seems to work fine in latest 3.4-140 (tested it for few times)

closed 2017-11-14 by Jackysi