Jackysi/advancedtomato #238: QoS Misclassification

farnwomt, 2016-10-10:
I have noticed an apparent fault with the QoS system.

When doing a large download I can go to the "Quality of Service->View Details" and I can see that the connection is classified as a FileXfer.

However when I go to "Quality of Service->View Graphs" I can see on the graph that the connection is classified as P2PBulk.

Furthermore it is quite clear that the bandwidth is being limited a though it is a P2PBulk connection and limited to 30% unlike a FileXfer which has a limit of 100%.

I am using 1.28.0000 MIPSR2-3.3-138 K26 USB Mega-VPN

Jackysi, 2016-10-11:
Sounds more like firmware issue than GUI. I will check where these things are defined, but I'm pretty sure its not in the GUI.

farnwomt, 2016-10-12:
Having looked a little more closely I think that whilst outgoing traffic is split up all my incoming traffic is being classified as P2PBulk.  Not sure why, but I am happy to report it elsewhere if you can point me in the right direction.

Thanks,
Michael

f00b4r0, 2016-10-24:
Hi,

I have the exact same problem, and I've identified the cause: in /etc/wan_qos, all IMQ IPV4 filters miss the "protocol ip" argument:

      # class id 90: rate 30kbit ceil 300kbit
      $TCA_IMQ parent 1:1 classid 1:90 htb rate 30kbit ceil 300kbit prio 9 quantum 1500
      $TQA_IMQ parent 1:90 handle 90: $Q
      $TFA_IMQ parent 1: prio 90 handle 9 fw flowid 1:90 
      $TFA_IMQ parent 1: prio 190 protocol ipv6 handle 9 fw flowid 1:90 

adding "protocol ip" in the first $TFA_IMQ line fixes the bug and makes inbound classifier work.

I don't know if that's an upstream bug, but at least I hope this info will help quickly fix it.

Jackysi, 2016-10-25:
https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-10-25 by Jackysi

farnwomt, 2016-10-25:
I really appreciate your efforts in locating this issue.  Obviously QoS 
is one of the big reasons to use Tomato, so having this fixed makes a 
big difference.

f00b4r0, 2016-10-25:
You're welcome. I have added extra information about this issue and a fix suggestion here: [bitbucket/tomato-arm:#74](https://bitbucket.org/pl_shibby/tomato-arm/issues/74/)

farnwomt, 2016-12-11:
Unfortunately this issue with the upstream code hasn't been fixed yet and it really does mean that AdvancedTomato doesn't work well for QoS if you are using the latest code.

The hack to fix it for now is to go to Administration->Scripts->Firewall

and to add this:

`
(
sleep 15
cat /etc/wan_qos  | sed 's/ prio 10 handle / prio 10 protocol ip handle /g' | sed 's/ prio 20 handle / prio 20 protocol ip handle /g' | sed 's/ prio 30 handle / prio 30 protocol ip handle /g' | sed 's/ prio 40 handle / prio 40 protocol ip handle /g' | sed 's/ prio 50 handle / prio 50 protocol ip handle /g' | sed 's/ prio 60 handle / prio 60 protocol ip handle /g' | sed 's/ prio 70 handle / prio 70 protocol ip handle /g' | sed 's/ prio 80 handle / prio 80 protocol ip handle /g' | sed 's/ prio 90 handle / prio 90 protocol ip handle /g' > /tmp/wan_qos
OLDMD5=`md5sum /tmp/wan_qos | cut -f1 -d' '`
NEWMD5=`md5sum /etc/wan_qos | cut -f1 -d' '`
if test "$OLDMD5" != "$NEWMD5"; then
    echo "Updated script."
    cp /tmp/wan_qos /etc/wan_qos
    chmod 700 /etc/wan_qos
    /etc/wan_qos stop
    /etc/wan_qos start
fi
) &
`

Yes this code is horrid and I am not proud of it at all, but it does the trick until somebody fixes the underlying issue.