Jackysi/advancedtomato #240: 2.4Ghz radio dissapears after router restart

jingcao, 2016-10-21:
Using Advanced Tomato version: 3.3-138 

On EA6500 V2, after restarting the router with both 2.4GHz and 5Ghz enabled, sharing the same SSID, the same security encryption, and password. The router boots up with only 5Ghz radio. GUI will no longer display 2.4Ghz radio after restart. 

To recover, wipe NVRAM and start from scratch (never restart). But after a restart the 2.4GHZ will be gone again.

Jackysi, 2016-10-21:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato by Shibby and so its not an actual firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-10-21 by Jackysi

jingcao, 2016-10-21:
The GUI is no longer displaying 2.4Ghz is a firmware issue?

Jackysi, 2016-10-22:
Yeah the firmware web handler takes care of the data provided to the UI. The UI displays array of wireless interfaces provided by the firmware. If that is missing, the issue occurs. Try Shibby's firmware, the only way to prove me wrong, but I sincerely doubt that would be the case.

jingcao, 2016-10-23:
Thanks, I tried Shibby's load and it also suffers the same problem.

Jackysi, 2016-10-23:
Yeah, thought so =). If you post to Shibby's issue's tracker maybe he'll fix it. I don't have the knowledge sorry.