Jackysi/advancedtomato #244: Can't change VLAN settings

elclaudio, 2016-10-24:
Hi

It seem's I can't remove a LAN port from the default VLAN 1 and affect it to another VLAN; well it appear to works in the gui, the router ask me to reboot and when I go back to VLAN advanced settings, the port I unchecked is still selected in VLAN 1, as well as the new wanted port...

in fact, I can't remove any LAN port from default VLAN, after reboot, the VLAN aren't changed from default configuration. NVRAM has been cleared after firmware installation.

Router model : netgear R7000
Advanced tomato version : Tomato v1.28.0000 -3.3-138 K26ARM USB AIO-64K

[image]

Jackysi, 2016-10-24:
This problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato by Shibby and so its not an actual firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open!

closed 2016-10-24 by Jackysi

elclaudio, 2016-10-24:
thanks
Btw, just tested, it works as expected with toastman's tomato build, tomato-R7000-9007.8Toastman-ARM-VPN-64K

Jackysi, 2016-10-25:
Toastman != Shibby / AdvancedTomato