Jackysi/advancedtomato #246: QOS: Settings for DSL only break QOS

f00b4r0, 2016-10-24:
When setting DSL overhead value to anything other than "None", the generated wan_qos script fails with errors and QOS is broken:

```
What is "atm"?
Usage: ... qdisc add ... htb [default N] [r2q N]
 default  minor id of class to which unclassified packets are sent {0}
 r2q      DRR quantums are computed as rate in Bps/r2q {10}
 debug    string of 16 numbers each 0-3 {0}

... class add ... htb rate R1 [burst B1] [mpu B] [overhead O]
                      [prio P] [slot S] [pslot PS]
                      [ceil R2] [cburst B2] [mtu MTU] [quantum Q]
 rate     rate allocated to this class (class can still borrow)
 burst    max bytes burst which can be accumulated during idle period {computed}
 mpu      minimum packet size used in rate computations
 overhead per-packet size overhead used in rate computations
 ceil     definite upper class rate (no borrows) {rate}
 cburst   burst but for ceil {computed}
 mtu      max packet size we create rate map for {1600}
 prio     priority of leaf; lower are served first {0}
 quantum  how much bytes to serve from leaf at once {use r2q}

TC HTB version 3.3
RTNETLINK answers: Invalid argument
RTNETLINK answers: Invalid argument
We have an error talking to the kernel
```

This is due to extraneous "atm" parameter added after the actual overhead value, which is not understood by tc. Removing this parameter fixes the problem.

The last two lines are due to the other bug reported here: #238 

Hope that helps!

Jackysi, 2016-10-25:
I don't modify non-gui files so this issue should be reported to Shibby. I sound like broken record...

closed 2016-10-25 by Jackysi

f00b4r0, 2016-10-25:
May I suggest you review the wording of this page:
https://advancedtomato.com/contact

and make it _very_ clear that people should only report to github GUI issues, and provide the link to Shibby's bitbucket for non-GUI issues? Reading the page it felt as if _all_ issues with Advanced Tomato should be reported here.

Also sometimes it's hard to tell which are GUI and which are non-GUI issues.

Cheers

Jackysi, 2016-10-25:
I don't understand how hard is it to understand that AdvancedTomato is a GUI not Firmware. That's the basic principle that any one should understand. Already from that stand point any issue related not directly to the GUI does not belong to its issue tracker.

I don't have the knowledge necessary to do that. Leaving all these issues opened up is just stupid since I can never fix them and the actual issues I can fix will be never fixed. 

https://advancedtomato.com/
*"Ahm, so what is AdvancedTomato?"*

f00b4r0, 2016-10-25:
> If you encounter errors or issues with AdvancedTomato it's best that you report them on Github repository under Issues tab. This will help **Tomato and** AdvancedTomato developers to fix it or to help you.

Just make it clear. Provide both links. People who download _your_ software don't necessarily know (or want to know) who did what. Make it clear on your website and you will get less noise. My 2c.

f00b4r0, 2016-10-25:
Apparently this bug is already fixed in Shibby's v138, but your code has the bug. See [#100](https://bitbucket.org/pl_shibby/tomato-arm/issues/100/qos-settings-for-dsl-only-break-qos)