Jackysi/advancedtomato #254: Bandwidth Limiting Default Class breaks connectivity completely

someinternetguy, 2016-11-11:
I'm having an issue with my AC66U where as soon as I enable any of the "Default Class for..." check boxes, all internet connectivity on my router is blocked.  I have several VLANs and would like to limit bandwidth of a few of them, but no matter which LAN I check off, it breaks connectivity for all of them. I think this is a bug... but any help would be appreciated.  thanks!

[image]

I'm on AC66U with AT v.1.28.0000 MIPSR2-3.3-138 K26AC USB AIO-64K freshly installed a few days ago.

Jackysi, 2016-11-11:
This is not "support" section. Rather "Issues Tracker", so please don't expect "help". Also this problem is related to Tomato router firmware, please report it to Shibby. AdvancedTomato is a GUI modification for Tomato by Shibby and so its not an actual firmware. Issues as previously stated are irrelevant to the Graphical user interface but are actual firmware issues. https://bitbucket.org/pl_shibby/tomato-arm/issues?status=new&status=open

closed 2016-11-11 by Jackysi

someinternetguy, 2016-11-11:
Thanks for the "friendly" response "Jackysi".  That was really "helpful".  If you read my "bug report", you would know that I think this is a "bug", but I'm just not sure.  Thought I would report here before going any further. Have a nice "day".

someinternetguy, 2016-11-11:
Sarcasm aside - I was linked here from your website. It specifically says to leave "errors or issues" here so "**tomato** and advanced tomato" developers can fix it or "**help**".  I am truly sorry if I reported this in the wrong place.  Although, you may want to change the link or wording on your website if so...
[image]

Jackysi, 2016-11-11:
The text is all right, if you would read home page (Takes about 5min total) you would notice that Advanced Tomato is GUI modification of Tomato by Shibby. GUI = Graphical User Interface. The issue above (as you stated) stops your connectivity. So how on earth would that be connected to the Interface that allows you to modify router options? In anyway its not possible that such options would disturb your connection. 

I'm sick of you people not reading basic principle of what my project is. I'm more sick of having to close 226 issues on which 200 are like this one. Completely unrelated to my project.

someinternetguy, 2016-11-11:
I do know the difference.  I know exactly what adv tomato is and is not.  Again, I don't know if the issue is just with adv tomato or underlying fw.  Maybe it is an issue with the GUI not properly communicating with the underlying fw...  I don't know, that's why I was letting you know. No need to be a d*ck about it.  

Your website text **is** misleading. The only reason I posted here first instead of direct to Tomato tracker (which was my original thought) is because of that wording.