Jackysi/advancedtomato #255: Spam relay hidden inside AdvancedTomato ?

Spheerys, 2016-11-13:
Hi !

I'm faced to a very strange issue.

I have a NETGEAR WNR3500Lv2 device, and since several month I had switched from TomatoUSB to AdvancedTomato

Last week, I have installed a [Pi-Hole](https://pi-hole.net/) to remove all ads on my network
Once pi-hole installed, I have checked the DNS resolutions of all my devices, by curiosity.
(Pi-hole is working very well :) )

That is at this moment that I have found those kind of suspicious DNS resolutions. 
Have a look : 

```
Nov 13 18:40:49 dnsmasq[439]: query[AAAA] mx-ha02.web.de from 192.168.0.254
Nov 13 18:41:24 dnsmasq[439]: query[AAAA] smtp.rgv.rr.com from 192.168.0.254
Nov 13 18:41:45 dnsmasq[439]: query[A] smtp.sina.com from 192.168.0.254
Nov 13 18:42:21 dnsmasq[439]: query[AAAA] smtp.163.com from 192.168.0.254
Nov 13 18:42:45 dnsmasq[439]: query[AAAA] smtp.newmail.ru from 192.168.0.254
Nov 13 18:43:34 dnsmasq[439]: query[AAAA] smtp.dodgeit.com from 192.168.0.254
Nov 13 18:43:57 dnsmasq[439]: query[AAAA] smtp.tiscali.it from 192.168.0.254
Nov 13 18:44:27 dnsmasq[439]: query[AAAA] mail.tin.it from 192.168.0.254
Nov 13 18:44:54 dnsmasq[439]: query[AAAA] smtp.rapidmobiles.co.uk from 192.168.0.254
Nov 13 18:45:37 dnsmasq[439]: query[AAAA] mail.oldskool.ru from 192.168.0.254
Nov 13 18:46:27 dnsmasq[439]: query[AAAA] smtp.buziaczek.pl from 192.168.0.254
Nov 13 18:46:49 dnsmasq[439]: query[AAAA] mxs.mail.ru from 192.168.0.254
Nov 13 18:47:43 dnsmasq[439]: query[AAAA] dnvrco-pub-iedge-vip.email.rr.com from 192.168.0.254
Nov 13 18:48:28 dnsmasq[439]: query[AAAA] smtp.rochester.rr.com from 192.168.0.254
Nov 13 18:48:59 dnsmasq[439]: query[AAAA] securesmtp.t-online.de from 192.168.0.254
Nov 13 18:49:20 dnsmasq[439]: query[AAAA] smtp.onet.pl from 192.168.0.254
Nov 13 18:49:41 dnsmasq[439]: query[AAAA] smtp.op.pl from 192.168.0.254
Nov 13 18:49:47 dnsmasq[439]: query[AAAA] ncfgraphics.com from 192.168.0.254
Nov 13 18:49:48 dnsmasq[439]: query[A] ncfgraphics.com from 192.168.0.254
Nov 13 18:50:07 dnsmasq[439]: query[AAAA] smtp.pochta.ru from 192.168.0.254
Nov 13 18:50:39 dnsmasq[439]: query[AAAA] pop.o2.pl from 192.168.0.254
Nov 13 18:51:12 dnsmasq[439]: query[AAAA] smtp.spoko.pl from 192.168.0.254
Nov 13 18:51:53 dnsmasq[439]: query[AAAA] smtp.vp.pl from 192.168.0.254
Nov 13 18:53:45 dnsmasq[439]: query[AAAA] smtp.tiscali.co.uk from 192.168.0.254
Nov 13 18:54:24 dnsmasq[439]: query[AAAA] smtp.pochta.ru from 192.168.0.254
Nov 13 18:55:00 dnsmasq[439]: query[AAAA] smtp.163.com from 192.168.0.254
Nov 13 18:56:15 dnsmasq[439]: query[AAAA] mx.autorambler.ru from 192.168.0.254
Nov 13 18:57:44 dnsmasq[439]: query[AAAA] smtp.parkmail.dk from 192.168.0.254
```

I have shutdown all devices on my network except the pi-hole (on a raspberry) and my Router Netgear (IP 192.168.0.254).
After one minute, the suspicious DNS resolutions continuing.

So I decide also to switch off my router. 
There, after 30 minutes, no more suspicious DNS resolutions !

After a restart of the router, the suspicious DNS resolutions comes back !

I say to myself : "oh sh****, my device was hacked !"
I ran an old version of AdvancedTomato (Netgear-3500Lv2-K26USB-1.28.AT-RT-N5x--3.0-132-AIO.chk)
So I have installed the very last version : Netgear-3500Lv2-K26USB-1.28.AT-RT-N5x--3.3-138-AIO.chk
I have checked the MD5 of course before flahing, and it was correct.

But after upgrade, the suspicious DNS resolutions were still there !

It's looking my device was transform to a smtp relay to spam !

So I asking to you : how can it be possible ? Is someone can reproduce this ?
If not, what did I miss ?

Spheerys, 2016-11-13:
Connected by ssh to my router, a `ps` command give me this : 
```
  PID USER       VSZ STAT COMMAND
    1 root      1404 S    /sbin/init
    2 root         0 SW<  [kthreadd]
    3 root         0 SW<  [ksoftirqd/0]
    4 root         0 SW<  [events/0]
    5 root         0 SW<  [khelper]
   21 root         0 SW<  [kblockd/0]
   24 root         0 SW<  [khubd]
   47 root         0 SW   [pdflush]
   48 root         0 SW   [pdflush]
   49 root         0 SW<  [kswapd0]
   50 root         0 SW<  [aio/0]
   92 root         0 SW<  [mtdblockd]
  385 root       656 S    hotplug2 --persistent --no-coldplug
  425 root      1388 S    buttons
  426 root      1356 S    console
  427 root      1628 S    /bin/sh
  535 root      1004 S    eapd
  538 root      1108 S    nas
  612 root      1116 S    dropbear -p 22 -a
  615 root      1004 S    eapd
  618 root      1108 S    nas
  631 root      1648 S    crond -l 9
  633 root       948 S    rstats
  637 root      1000 S    cstats
 1053 root      1636 S    udhcpc -i vlan2 -b -s dhcpc-event -H tomato -O 33 -O 121 -O 249 -m -p /var/run/udhcpc-wan.pid
 1058 nobody    1656 S    dnsmasq -c 4096 --log-async
 1346 root      3132 S    httpd
 1354 root      1180 S    dropbear -p 22 -a
 1355 root      1180 S    dropbear -p 22 -a
 1789 root      1180 S    dropbear -p 22 -a
 3082 root      1180 S    dropbear -p 22 -a
 4810 root      1180 S    dropbear -p 22 -a
 5611 root      1184 S    dropbear -p 22 -a
 6145 root      1256 S    dropbear -p 22 -a
 6146 root      1632 S    -sh
 6571 root      1180 S    dropbear -p 22 -a
 6572 root      1644 S    -sh
 6722 root      1624 R    ps
```
The multiple dropbear commands surprised me, especially since on the [official dropbear website](https://matt.ucc.asn.au/dropbear/dropbear.html) they are talking about vulnerability found on july 2016
But a `dropbear -V` give me `Dropbear v2016.74`, so I suppose it is OK...

plfort, 2016-11-14:
Do you allow SSH access from WAN ?
Did you try a netstat -a ?

Spheerys, 2016-11-14:
I thought it was desactivated (because I don't need it), but it was switched on !
And a `netstat -a` is very clear !!!

```
# netstat -a
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 tomato:www              0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:domain          0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN      
tcp        0      0 tomato:ssh              192.168.0.3:56506       ESTABLISHED 
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh 62-210-105-222.rev.poneytelecom.eu:3050 FIN_WAIT2   
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh 45.63.77.89.vultr.com:55416 TIME_WAIT   
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh static.92.140.201.138.clients.your-server.de:49456 FIN_WAIT2   
tcp        0      1 *****my-private-reverse-dns********.proxad.net:ssh dedic803.hidehost.net:24899 FIN_WAIT1   
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh vk6417149.myihor.ru:56779 FIN_WAIT2   
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh ns359440.ip-91-121-148.eu:49932 FIN_WAIT2   
tcp        0      1 *****my-private-reverse-dns********.proxad.net:ssh vk6417149.myihor.ru:52635 FIN_WAIT1   
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh 68.64.168.202:64737     FIN_WAIT2   
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh 82.200.204.254:54660    FIN_WAIT2   
tcp        0      0 *****my-private-reverse-dns********.proxad.net:ssh 109.105.59.22:50855     FIN_WAIT2   
tcp        0      0 :::domain               :::*                    LISTEN      
tcp        0      0 :::ssh                  :::*                    LISTEN      
udp        0      0 localhost:38032         0.0.0.0:*                           
udp        0      0 localhost:38032         0.0.0.0:*                           
udp        0      0 0.0.0.0:domain          0.0.0.0:*                           
udp        0      0 0.0.0.0:bootps          0.0.0.0:*                           
udp        0      0 0.0.0.0:38000           0.0.0.0:*                           
udp        0      0 0.0.0.0:38000           0.0.0.0:*                           
udp        0      0 :::domain               :::*                                
raw        0      0 0.0.0.0:255             0.0.0.0:*               255         
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node Path
unix  2      [ ]         DGRAM                     23543 

```
I have switched off the remote ssh access, but there are still 2 c

plfort, 2016-11-14:
Try to kill all the dropbear processes:
kill 1354 1355 ...

plfort, 2016-11-14:
You should also check for the file authorized_keys in /tmp/home/root/.ssh, remove offending public keys from this file !
Disallow password authentication too

octo-byte, 2016-11-14:
Attempted to reproduce, but couldn't really make a conclusive result.

### Versioning
```
Netgear R6250
Tomato Firmware 1.28.0000 -3.3-138 K26ARM USB AIO-64K
Dropbear v2016.74
```

### ps report
```
PID USER       VSZ STAT COMMAND 
    1 root      1608 S    /sbin/preinit 
    2 root         0 SW   [kthreadd] 
    3 root         0 SW   [ksoftirqd/0] 
    4 root         0 SW   [kworker/0:0] 
    5 root         0 SW   [kworker/u:0] 
    6 root         0 SW   [migration/0] 
    7 root         0 SW   [migration/1] 
    8 root         0 SW   [kworker/1:0] 
    9 root         0 SW   [ksoftirqd/1] 
   10 root         0 SW<  [khelper] 
   11 root         0 SW   [kworker/u:1] 
   53 root         0 SW   [sync_supers] 
   55 root         0 SW   [bdi-default] 
   56 root         0 SW<  [kblockd] 
  111 root         0 SW   [kswapd0] 
  158 root         0 SW   [fsnotify_mark] 
  166 root         0 SW<  [crypto] 
  242 root         0 SW   [mtdblock0] 
  247 root         0 SW   [mtdblock1] 
  252 root         0 SW   [mtdblock2] 
  257 root         0 SW   [mtdblock3] 
  262 root         0 SW   [mtdblock4] 
  293 root         0 SW   [kworker/0:1] 
  294 root         0 SW   [kworker/1:1] 
  297 root         0 SW   [mtdblock5] 
  302 root         0 SW   [mtdblock6] 
  707 root       664 S    hotplug2 --persistent --no-coldplug 
  750 root      1592 S    buttons 
  752 root      1596 S    blink_5g 
  753 root      1528 S    console 
  755 root      1376 S    syslogd -L -s 50 -b 1 
  756 root      1384 S    /bin/sh 
  758 root      1376 S    klogd 
  774 root         0 SW   [khubd] 
  839 root       656 S    sd-idle 
  841 root         0 SW<  [usbhid_resumer] 
  964 root      1036 S    eapd 
  966 root      1592 S    nas 
 1037 root      1376 S    telnetd -p 23 
 1041 root      1056 S    dropbear -p 22 
 1056 root      1036 S    eapd 
 1059 root      1592 S    nas 
 1215 root      1404 S    crond -l 9 
 1219 root      1060 S    rstats 
 1267 root      1108 S    cstats 
 1271 root      3752 S    nginx -c /tmp/etc/nginx/nginx.conf 
 1310 root      2376 S    nmbd -D 
 1311 root      2308 S    nmbd -D 
 1312 root      3060 S N  smbd -D 
 1323 root      1600 S    blink eth1 wlan 20 8192 
 1330 root      1600 S    blink eth2 5g 20 8192 
19294 root      1424 S    /tmp/ppp/pppdwan file /tmp/ppp/wan_options 
25891 nobody   14124 S    dnsmasq -c 4096 --log-async 
29699 root      2536 S    miniupnpd -f /etc/upnp/config 
29714 root      2816 S    httpd 
31137 root      2828 S    httpd 
31138 root      1380 S    sh -c /tmp/.wxDeT65s 2>&1 
31139 root      1380 S    {.wxDeT65s} /bin/sh /tmp/.wxDeT65s 
31143 root      1380 R    ps
```

My logs don't report any dnsmasq queries to the given domains. I am unable to resolve the DNS for any of the AAAA records as I haven't configured IPv6, however I did test the A records. ```ncfgraphics.com``` did not resolve, but ```smtp.sina.com``` did to 202.108.6.242.

Spheerys, 2016-11-14:
After desactivated "Remote Access" and "Remote Forwarding" and restarted the router, a netstat -a is more appreciable : 
```
# netstat -a
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 tomato:www              0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:domain          0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN      
tcp        0      0 tomato:ssh              192.168.0.3:57334       ESTABLISHED 
tcp        0      0 :::domain               :::*                    LISTEN      
tcp        0      0 :::ssh                  :::*                    LISTEN      
udp        0      0 localhost:38032         0.0.0.0:*                           
udp        0      0 localhost:38032         0.0.0.0:*                           
udp        0      0 0.0.0.0:domain          0.0.0.0:*                           
udp        0      0 0.0.0.0:bootps          0.0.0.0:*                           
udp        0      0 0.0.0.0:38000           0.0.0.0:*                           
udp        0      0 0.0.0.0:38000           0.0.0.0:*                           
udp        0      0 :::domain               :::*                                
raw        0      0 0.0.0.0:255             0.0.0.0:*               255         
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node Path
unix  2      [ ]         DGRAM                      2191 

```

So I suppose my mistake comes from this remote access... I'm very confused :S 
My pihole don't record any suspicious DNS request... so it's looking good...
I will re-flash my firmware, just in case...

octo-byte, 2016-11-14:
Yeah good move. Glad you were able to sort it out before it was too late, ahah

plfort, 2016-11-14:
I am worry about this remote access ... is your ssh password weak ?

Spheerys, 2016-11-14:
I'm shamed to confess...
Default passwd + remote access = boom

I don't understand how I could keep this remote access on !
And I didn't changed the default password because I thought protected... tsssss bullshit....

closed 2016-11-14 by Spheerys

plfort, 2016-11-14:
Ok cool (well, i mean not so cool for you but it's pretty reassuring)

Jackysi, 2016-11-14:
I apologize for not anticipating in this issue report. I didn't really know how to respond. Happy you solved the issue!

Spheerys, 2016-11-14:
Don't worry, I can understand :) 
I'm happy the issue is closed now