Jackysi/advancedtomato #256: Search config options by keyword

doublerebel, 2016-11-16:
Recently a user in #tomato was wishing for a search feature.  I think this would be reasonably easy to add to AdvancedTomato, since AdvancedTomato is all clientside.  It would be ideal to search by nvram config name as well as GUI label, and get a link to whatever page/setting is appropriate.  I am a JS dev so I have an idea how that could work.

Thanks for your work, AdvancedTomato is great.

Jackysi, 2016-11-16:
I was thinking about that too! However without some special work I'm not sure how that would work because Javascript has no access to "list" of pages/their content unless its already loaded once. If you have an idea, that would really help going towards this functionality.

I was also thinking to add notifications system and put restarts and warnings etc there, just a thought.

doublerebel, 2016-11-16:
I think the easiest way would be to use a pre-processor to scan all the pages and create a dictionary/map of settings/labels to pages.  Since AdvancedTomato doesn't change once it's deployed, there's no need for JS to have to search 'live', the map can be created as part of the build process.

Then we just use some JS clientside autocomplete/suggestion module to handle searching the dictionary/map.

Jackysi, 2016-12-02:
Sorry for very long delay in response, been very busy. You mean like creating one index file with something like JSON list of every thing. That would work but size of such file would be huge :). Idea is awesome but implementation is rather different story. We would also have to pay attention to enabled/disabled features (not all firmware images are AIO). 

Its doable but not with my knowledge, I could do front-end but the indexing work I don't know C language. To parse all pages and put them into file. I'm PHP web developer :(.

doublerebel, 2016-12-02:
No worries, plenty to do around here too :)

I don't think we need to do it in C, we could probably do it in Bash.  It would have to be part of the final build step so that it knows the enabled/disabled features for that specific build.  I might have some time next week to make a proof-of-concept, and we could see how large the index files would be.