Jackysi/advancedtomato #262: RT-N16 crashes with WPA2 Personal AES enabled

olind, 2016-12-02:
Hi,
I've just installed _tomato-K26USB-1.28.AT-RT-N5x-MIPSR2-3.3-138-AIO.trx_ on my _RT-N16_ and have used it for some days. I reset NVRam carefully and did only change WiFi encryption to WPA2 Personal AES. Tested to download some files (http://www.thinkbroadband.com/download.html) and the router crashes.

I tried with both _tomato-K26USB-1.28.RT-MIPSR2-137-AIO_ & _tomato-K26USB-1.28.RT-MIPSR2-138-AIO_

Are I doing anything wrong or is there an issue?

Can I do anything more to help find the problem? 

Are there any workarounds?

Thanks in advanced

Jackysi, 2016-12-05:
I would say this is firmware issue which report goes to Shibby. Also I'd say Tomato is one of most stable firmwares available, so its more likely your router's hardware is responsible for crashing than tomato it self.

closed 2016-12-05 by Jackysi

olind, 2016-12-11:
You were right. Thanks for pointing that out.