Jackysi/advancedtomato #268: "Update available!" message even after already updated

vrapp, 2016-12-06:
After firmware was updated to 3.4-138 , it still shows "Update Available! AdvancedTomato version 3.4-138 has been released and it is available for download.	"
[image]

Jackysi, 2016-12-06:
Which router? It could be cache which is strange because tmp folder clears on reboot (after flash). I have N66U and R7000 and I can't replicate the issue.

vrapp, 2016-12-06:
Asus RT-AC68P/U B1

The message showed up immediately after having upgraded. Then I clicked "x" to close it, it closed and no longer shows up.

vrapp, 2016-12-06:
...which ,maybe presents another problem. If this message can be hidden by closing it, if the upgrade _was_ in fact available, how to re-invoke it again? There's no "check for upgrade" button on the "upgrade" page admin-upgrade.asp .

Jackysi, 2016-12-06:
When you hide update box, it hides that specific update so 3.4-138. Next update 3.5-138 or 3.4-139 will be shown normally.

vrapp, 2016-12-06:
What if I hide the next update without installing it, for whatever reason? will I be able to see it again?

Jackysi, 2016-12-06:
The specific version you disabled will not ever appear again. But if in above example 3.5-139 is released, you will see the notification again.

Jackysi, 2017-11-14:
I can't reproduce this issue. Been trying to since version 138 :/

closed 2017-11-14 by Jackysi