Jackysi/advancedtomato #269: Unchecked checkboxes are invisible

vrapp, 2016-12-06:
With the current color scheme. unchecked checkboxes are almost invisible until you hover the mouse over them. Here's typical example.

[image]

[image]

Tried in Firefox, Chrome, and IE - looks the same everywhere.

Jackysi, 2016-12-06:
They're supposed to be. If using RGB or sRGB monitor it should appear just slightly. Which is the idea because you can also click on "Enable" and it will check the check box. Also not sure what this is? bug report defo since this is not a bug =)

closed 2016-12-06 by Jackysi

vrapp, 2016-12-06:
The problem with clicking on "enable" is that you don't see where to click. Really, if I need to enable something, I have to slowly move the mouse horizontally until the cursor changes, that's how I identify where the checkbox is. When I used it first time, I could not understand how to enable things at all, until I finally figured, almost accidentally, that in fact there's a checkbox, although almost invisible. Be at just a slightly wrong angle to the monitor, and you already don't see the checkbox, at least not without really, really closely looking. It shouldn't be that way.

As for community forums - if I understand correctly, advanced tomato is exactly about web interface, no? I rather had doubts about the b.r. about the certificate belonging to advancedtomato.

Jackysi, 2016-12-06:
Yes, but that's up to a discussion. I don't agree with you for example. Unless at least some amount of people agree with you, I won't implement changes to that. That's not a bug its how I wanted it. This is still my project so its up to me how the things are going to look like no?

 I don't have the same issues as you, guess quality monitor (I have 3 calibrated monitors) make a difference.

vrapp, 2016-12-09:
[One more user agreed](http://www.linksysinfo.org/index.php?threads/advancedtomato-flat-material-designed-web-gui-shibby-base.70254/page-9#post-282259) :-)