Jackysi/advancedtomato #274: Connection lost error

Rel1k, 2016-12-15:
v3.4-138Max

When trying to access the "wireless" section under "Advanced Settings" the loading bar gets about half way through loading and then states "Connection lost error" and I am unable to get to that section. I tried different computers and browsers with the same results.

Jackysi, 2016-12-25:
Check F12 developers console and paste the screen shot if any errors occur. I'm pretty sure there is something wrong with your NVRAM values or router configuration which is causing javascript errors which end up showing "Connection Lost Error".

Mako88, 2017-04-06:
[image]

I'm getting the same error, and here's a screenshot :)

Jackysi, 2017-04-07:
From screenshot above its visible that router simply does not respond. The UI it self doesn't really have anything to do with that. Is there anything about HTTPD errors in Logs?

garzonm77, 2017-04-22:
I had the same issue clicking on upgrade. As a workaround, I was able to right-click on Upgrade and open in New Tab...it opened up and upgraded to the latest version for my router (Netgear r7000). After the upgrade I didn't have the issue anymore.

Jackysi, 2017-05-08:
Guess its resolved? No one bothers to write about it anymore =)?

closed 2017-05-08 by Jackysi