Jackysi/advancedtomato #275: OUI lookup - dashes in MAC binring incorrect results

vrapp, 2016-12-17:
When I use shortcut "OUI" to look up the MAC of my Vizio TV, beginning with 00:6B:9E, the resulting page  https://services13.ieee.org/RST/standards-ra-web/rest/assignments/download/?registry=MA-L&format=html&text=00-6B-9E has multiple entries, none of which is Vizio. The same page correctly returns Vizio if we remove the dashes from the URL, i.e. https://services13.ieee.org/RST/standards-ra-web/rest/assignments/download/?registry=MA-L&format=html&text=006B9E

Jackysi, 2016-12-25:
I need to find how this function works and why slashes are there

Jackysi, 2016-12-25:
I just tested some other devices and I get 500 error when using without slashes. Your solution might not be really a solution at all.

https://services13.ieee.org/RST/standards-ra-web/rest/assignments/download/?registry=MA-L&format=html&text=50465D
vs
https://services13.ieee.org/RST/standards-ra-web/rest/assignments/download/?registry=MA-L&format=html&text=50-46-5D

or

https://services13.ieee.org/RST/standards-ra-web/rest/assignments/download/?registry=MA-L&format=html&text=C0-EE-FB
vs
https://services13.ieee.org/RST/standards-ra-web/rest/assignments/download/?registry=MA-L&format=html&text=C0EEFB

In some cases removing dashes works in some not. At the time I will say this is IEEE Standards issue which we can't fix by simply changing dashes in the search string. So this issue is not related to Tomato.

closed 2016-12-25 by Jackysi