Jackysi/advancedtomato #278: ER: Download Manager

vrapp, 2016-12-24:
It would be nice to have HTTP/FTP download manager, in addition to torrent client. 

According to https://trac.transmissionbt.com/ticket/3410 , there's significant demand for such a feature - a number of users asked for it to be implemented in Transmission, but developer's answer was "over my dead body".

PyLoad would be nice addition to the firmware.

Jackysi, 2016-12-24:
Yea I don't think that will ever happen on Tomato it self. Not by me any how. I don't see usefulness in doing that on router. Even Torrent client on router is bit too much for me. I use VPN edition of the firmware =). I'm not saying I don't support the idea, I'm just saying I have better ways of using my time for which I believe is more essential

vrapp, 2016-12-25:
Not to argue, it's your product, but usefulness of torrent and download clients is in ability to download very large files, that takes hours and hours, without keeping computer on. The router is good choice for that also because it likely has USB drive with some terabytes, serving like family storage, and also because the same router is likely to very soon stream just-downloaded video file with its DLNA media server running on the same router. So it's very convenient to have everything related on one device.

Jackysi, 2016-12-25:
I have 500mbps connection. I downloaded 400GB file last week in just about 2hours. Regarding DLNA I find it the most useless function ever existed. I have many other ways to play files on TV and I have 2 TVs at home one 128cm and one 150cm so I won't watch movies on a table or PC. I mostly watch movies with subtitles and PotPlayer with SVP (making movies smooth 60fps) which makes purpuse of DLNA well, you know :). My Denon AVR-X2100W doesn't like DLNA at all (doesn't want to work, not sure why not, but since I prefer using other methods to play over LAN I never wasted my time on it).

I have very high performance Computer and Its shutdown only 8 hours a day. It has 8Tb space from which 1tb is SSDs. My router has 200gb drive which is slow (40mb read speed)... If I had a choice and possibility to go 10gbps LAN I would. 

See no every one uses the same functions and stuff. For example I use Nginx as proxy on my router, Bind DNS server and few other stuff like NodeJS for some personal projects. On my router regarding Tomato features I don't use FTP, DLNA, VPN (I never used that in my life, If i wanted to secure my internet I used PTPP), Tor (seriously I'm not a criminal needing such slow and safe network, not selling drugs or weapons online), Torrrent and many others. 

Don't get me wrong, I'm not trying to argue with you. I'm just explaining my point of view so perhaps you can get a feeling why features we are talking about feel useless to me. 

P.S.: I did read topic you sent above about the torrent client implementing FTP/HTTP downloads. I sort of agree with the author. But feel the requirement of certain people. So the solution is (if I remember correctly the project is open source) for someone who shares that idea to write that additionally and fork the project. Simple as that. I had to fork my project too because none of Tomato developers wanted new GUI.

vrapp, 2016-12-25:
Thanks for so detailed answer, very very interesting, even eye-opening, really appreciate that. But it probably begs the question: are open-source software authors making their products for themselves, or for others?

Jackysi, 2016-12-25:
I never said AdvancedTomato GUI project is for others. Heh I don't think all open source authors are doing products for others. I'm not. My project started for me and me only. I share it with others who may like it, for free without any warranty or terms, but I never said I will implement any idea and ideology they may have. I accept ideas that I see beneficial to every one but when people goal at their own needs or specifics I know for fact in future world won't be needed, I decline them. The world is moving forward not backwards that's also one of reasons I'm dropping support for older routers (and yes I understand not every one makes enough money to afford going forward or has ability to do so (other limitations)). I'm not rich my self, I waste money on technology because I love it. I drive a shit car and live in a crappy old house for that. Every one has its own priorities. For those who don't want to afford better router they can still use old firmware and original one.

Also you wrote above that "many people" wanted that feature, I saw only 16 comments. That's kind of rhetorical since Advanced Tomato is now used by 5500 people + (that's only the amount of people who share statistics, many have it disabled that's proven by 18,974 unique downloads on previous version). You can't really say ALL of them have EXACTLY same needs as we do (me and you). https://advancedtomato.com/statistics

I don't mind explaining my views on the world and projects so no problem :).

vrapp, 2016-12-25:
Of course it's not all, but since the ratio of those who (1) found the website where issues are discussed, (2) found the topic where it's discussed, (3) found the time and energy to post, after (4) having found it to register at the website, is probably very small, so with that in mind, 16 is a huge number :-) 

Same as even one of them who agreed that the borders of checkboxes and textboxes might be more contrast - that one is not huge, but "significant" :-)

99.9% of people who face some problem (not only with computers) simply go around, rather than engage in a effort to fix it.