Jackysi/advancedtomato #284: WAN DHCP option should be disabled for when on PPPoE mode

ssbarnea, 2017-01-06:
If WAN DHCP option is checked when WAN is in PPPoE mode port forwarding will not work. This is documented at http://samuelpinches.com.au/blab/port-forwarding-and-upnp-on-advanced-tomato-not-working/

See http://take.ms/LRUlr -- it should not be possible to enable DHCP on WAN with PPPoE in order to avoid this issue.

The GUI is confusing because on PPPoE the server decides if you get static or dynamic so DHCP option doesn't really matter. 

I do confirm that disabling the DHCP option and rebooting the router solves the problem. It would worth mentioning that I had to trigger a reboot from the console because doing it from the web interface didn't restart the router (was doing nothing.) but this is probably another bug unrelated to this.

pinchies, 2017-03-31:
Thanks for logging this. I tried mentioning it at (http://tomatousb.org/forum/t-1650384/n66u-shibby-v135:stuck-at-renewing-wan-dhcp) but it wasn't addressed. Hopefully it can be fixed now.

ssbarnea, 2017-03-31:
Sadly I sold my router and I am still missing AdvancedTomato, the best UI for a router so far. Sadly is supported on a very limited range of devices.

Jackysi, 2017-11-14:
I think this was fixed by 140, if not please let me know and I will reopen the issue.

closed 2017-11-14 by Jackysi

dewiestr, 2018-02-04:
140 did not fix this issue for me... Can I supply any logs ?

jump, 2021-09-24:
This is definitely still broken. I'm on latest AdvancedTomato in 2021 and both CTF being enabled and also enabling that DHCP option in PPPoE settings will break port forwarding. 

We should be hiding that DHCP option if PPPoE is selected. 

Wireless Client Mode as an option should also disappear if PPPoE is selected.