Jackysi/advancedtomato #288: Issue assigning one of my CAT5 ports to new vlan

rahimlila, 2017-01-22:
I'm running Tomato Firmware 1.28.0000 -3.4-138 K26ARM USB AIO-64K on my Asus Router (RT-AC68U)

I am having difficulty getting  port 4 on router to be assigned to a new vlan. vlan itself is fine and works through wifi. I first uncheck port 4 for br0, then I check it for br1. At which point I save settings and it reboots the router. But when the router gets back up, I see that the port is assigned to both bro and br1 in the UI. 

I've repeated the above steps 6 times and still get the same issue

rahimlila, 2017-01-22:
[image]

Here is a screenshot of Port 3 showing in the UI assigned to both vlans after router rebooted

dddango, 2017-01-27:
You have to setup VLAN with terminal connection. VLAN GUI has a bug and cannot change the settings.

So, let's connect to router with ssh, then type in the following:
Check the current VLAN: nvram show|grep vlan.ports
Set desired ports : nvram set vlan()ports="() () () ()"
Set manual boot: nvram manual_boot_nv=1
Write the changes: nvram commit
reboot the device: reboot

Just replace () with desired number. After you have rebooted VLAN should show up right. I had same problem and this fixed it for me.

rahimlila, 2017-01-27:
I flashed my router with tomato by shinny and solved the issue that way. Think I'm gonna just stick with shinny.

Thanks for responding. Hopefully bug is fixed soon

webmaster33, 2017-01-27:
Advanced Tomato is based on Tomato by shibby...

Yuki-nyan, 2017-03-19:
But the VLAN GUI works fine in shibby's builds...

simonz21, 2017-04-03:
I used the shibby builds and I have this exact same problem. It's a shibby problem, not an AT problem.

tavinus, 2017-04-22:
This seems to be related:
 - http://www.linksysinfo.org/index.php?threads/vlan-mess.72648/
 - http://www.linksysinfo.org/index.php?threads/tomato-shibby-138-vlans-not-working.73121/#post-283244  
 
They offer some potential fixes for v138 there.  
 
From what I read the problem is not present on Shibby's version 136. 
So AT's Version 3.2-136 may be fine too.  

I can't really test to confirm right now because I am waiting for my router.

Jackysi, 2017-05-08:
Version 140 will be released soon, is this issue still there?

closed 2017-05-08 by Jackysi