Jackysi/advancedtomato #291: DNSCrypt not working at power up because of TIME not set

agapecorporation, 2017-02-03:
- I have DNSCrypt working fine except that if the router loses power then it cannot set TIME as (DNS does not resolve).
- DNSCrypt will not initiate certificate because TIME is not set.
- All I have to do is uncheck the DNSCrypt option, let the TIME get set from NTP.org, Check DNSCrypt option and it is good to go again (until the next power cycle...

I am using a Linksys E2500V2 with Advanced Tomato v34-138 Max Firmware.
Please let me know if you need any more info.

I will be glad to make a donation to the project if this can be resolved.
Thanks!
Brad H

Jackysi, 2017-02-04:
When someone offers donation if their issue is resolved I intentionally leave that specific issue THERE and never ever take those people serious. Be my guest switch to another firmware.

closed 2017-02-04 by Jackysi

agapecorporation, 2017-02-04:
Terribly sorry I meant no offense!!  I was only offering to help, I am clueless to be able to make any changes myself as I am not a programmer.
Do you know of someone that would be interested in helping me? (for money or ??)
bh

Jackysi, 2017-02-04:
It sounds offensive when someone comes ahead with (I will be glad to make a donation to the project if this can be resolved.) because that sounds like: "Hey this project is great, but hey u don't deserve Donation unless you fix this specific issue.". We, developers, spent days, months, years perfecting this software/firmware for FREE. Offering donation just in case "that happens" is really offensive.

I don't know anyone. It looks like Shibby's has given up too. I doubt this is just being busy because hes been away for months. I won't draw up conclusions but Tomato doesn't have thriving future. The way some people behave its just, sad.

agapecorporation, 2017-02-04:
Please understand I am not from the open source community so please forgive if I come across offensive.  I will just put by cards on the table.  I am a business man and I see an opportunity with the DNSCrypt option.  This is a somewhat new dynamic in the world of security but I see it as something that is going to catch on.  Eventually all routers will have an Encrypted DNS option as standard (I think) but UNTIL then I see a small window where a little guy could sell Tomato routers preconfigured for DNSCrypt.
  I am not asking for anything free I am willing to negotiating payment to get the change, I am willing to talk about getting a custom version that would be licensed for each copy I use.  If this does not interest you i understand feel free to say so, I will keep looking.

Brad H

Jackysi, 2017-02-04:
Its okay.

I don't have the knowledge or time to mess around DNSCrypt. But If I were you I would post your idea on Linksysinfo forums (well, not exact idea, but the requirements + payment as in freelancing or paid feature not donation :=)) and see what happens.

I'm web developer so my work on Tomato is graphical only and my work at the moment takes 12h a day from Monday to Sunday.