Jackysi/advancedtomato #293: Daily/weekly bandwidth history not showing all WAN traffic

o355, 2017-02-11:
I'm having this really odd quirk with AdvancedTomato, so I'd like to know if it's something with the base firmware itself, or with the GUI. Not sure, but I'll report it here for now.

Basically, AdvancedTomato doesn't count ALL traffic coming from my many interfaces (eth1/eth2/wl0.1/wl0.2 (under br1)/wl1.1) to the daily/weekly traffic history. I've downloaded large Steam games, or done large downloads, and the bandwidth history does not paint the whole picture to my network's bandwidth history.

Last night, AdvancedTomato reported that my entire network used 0.11 GB, but that's incorrect, as I was downloading ISO files (that were more than 0.11 GB), and others in the house were streaming video.

Furthermore, I don't know if Tomato counts that usage from the WAN (vlan2) interface, but if it does, that's also completely off, and it doesn't display true WAN usage (br0 does, as that's where all traffic from wired/wireless feeds into).

Again, I don't know if this is an issue with Tomato itself, or AdvancedTomato's modifications. If this isn't an issue with AdvancedTomato, it'd be great if I could get led somewhere so I can report this issue to the proper people to get it fixed.

Jackysi, 2017-02-12:
AdvancedTomato has no modifications other than GUI. GUI only displays data it gets from Firmware. Basically I'm sure this is Tomato issue but for the doubt of my knowledge I'll leave this issue as it is.

SanderSchutten, 2017-03-27:
Do you have Cut-through-forwarding enabled?
Do you have QOS services enabled?
I think these options interfere with the monitoring of the traffic.

o355, 2017-03-29:
CTF is on, QOS is off.

Jackysi, 2017-05-08:
Its highly likely CTF is the issue here, unless proven otherwise I will close this issue.

closed 2017-05-08 by Jackysi