Jackysi/advancedtomato #295: Support Cloudflare in Dynamic DNS

overcookedTOFU, 2017-02-12:
Cloudflare has updated their api that allows one to change your dns configuration.  Tomato should support it.

I am not certain how Tomato currently does dynamic dns updates, but Cloudflare is supported in the newest version of ddclient (3.8.3).

Resources: 
https://support.cloudflare.com/hc/en-us/articles/200168816-Does-Cloudflare-work-with-Dynamic-DNS-Can-I-update-my-DNS-records-remotely-

https://www.cloudflare.com/technical-resources/#ddclient

Jackysi, 2017-02-12:
I don't do firmware stuff. Talk to Shibby about that ^^

closed 2017-02-12 by Jackysi

overcookedTOFU, 2017-02-12:
Will do.  I added a Feature Request here: http://tomato.groov.pl/?page_id=334&bugerator_nav=list&bug_project=1

Is that the right bug tracker?

Thanks!

Jackysi, 2017-02-14:
Bitbucket is better, its linked on my AdvancedTomato contact page

overcookedTOFU, 2017-02-14:
Thanks!  For future reference, here is the bug at Bitbucket: https://bitbucket.org/pl_shibby/tomato-arm/issues/133/support-cloudflare-in-dynamic-dns