Jackysi/advancedtomato #298: openVPN client silent fail

jontis, 2017-02-14:
I have an openVPN client running that seems to be working fine, traffic is routed with encryption to the VPN. Once in a while, this openVPN client fails, or something else in the system fails.
Traffic is then released directly out without encryption and routing and this happens silently. I only notice by checking my final IP address. This is of course horrible that the traffic can without notice bypass all intended security.
Restarting the openVPN client normally makes the encryption and routing work again.

The router should block traffic when the VPN fails, and maybe restart VPN
Is this possible?

closed 2017-02-14 by Jackysi

jontis, 2017-02-14:
Thanks for quick action, but could you just briefly comment on why this is not an issue here?
I noticed your gui is great for showing stats. Is working vpn not a stat? 
And if it is and issue for someone else, please indicate who? openVPN? Tomato firmware.

Many thanks for the great work.

Jackysi, 2017-02-14:
https://github.com/Jackysi/advancedtomato/issues/260
+200 more like it.